samyk/poisontap #2: osx mitigations?

kristovatlas, 2016-11-16:
Does anyone have any mitigations to prevent usb-ethernet adapters from being automatically connected to?

I've noticed the following behavior:
-If I create a new Network Location in the "Network" section of MacOS's System Preferences and the usb-ethernet adapter is plugged in, the new Location will automatically add the device to the list of adapters and connect to it.
-If I create a new Location without the device plugged in, activate that location, and then plug in the device, it will not be automatically added to the list of devices, nor automatically connected to.

memiux, 2016-11-16:
[image]

neuhaus, 2016-11-17:
Just in case someone plans to do this: How do you charge your new Macbook or Macbook Pro after you cemented the USB C ports :battery:  ?

JarrettR, 2016-11-17:
The new Macbooks support Wave wireless charging, don't they? Better try it to find out.

gddeen, 2016-11-17:
I would think if someone had access to a USB port, they might be clever enough to have
a screwdriver and design access directly from the board.

Security has always suggested no physical access.

JonathanNathanson, 2016-11-19:
https://www.amazon.co.uk/dp/B000FKP8B6/ref=twister_B01BP55BU2?_encoding=UTF8&psc=1

Potentially a better mitigation than cement, although not foolproof.

AlexPoulsen, 2016-11-21:
[https://derflounder.wordpress.com/2014/02/12/power-nap-power-management-settings-and-filevault-2/](https://derflounder.wordpress.com/2014/02/12/power-nap-power-management-settings-and-filevault-2/)

This walks you through getting your mac to hibernate instead of sleep. It is a little bit quicker than a full shutdown (Not tested, just felt quicker) and saves your ram so everything is where you left it.
Requires FileVault and copying and pasting commands into Terminal

Used a midi keyboard with a light on it to test if it was connected. Before changing the settings, On, after changing them, Off. So most likely this fixes it completely. Just expect to wait a good bit longer when waking it up. Uses hibernation for blocking access to the USBs, and FileVault so they can't just connect your mac up to another computer and access it's storage. Or manually take your drive out if you still have a mac that old.

_Way_ better than cement.

kristovatlas, 2016-11-21:
@AlexPoulsen I think that should help, but I'd like to see someone test it with PoisonTap. Note that if an attacker wakes your machine up while hibernated, he will be perfectly able to connect USB devices including USB-to-ethernet adapters. The reason I think it will still work decently is that, while the OS may connect to the USB-to-ethernet adapter via DHCP, presumably all of the apps that would send interesting data are not running until the hibernate image is decrypted and loaded back into RAM.

kristovatlas, 2016-11-21:
@JonathanNathanson I think that raises the bar very slightly, but those are pretty easy to remove with a small tool -- not to mention one of the tools the company itself sells if the attacker is expecting the port blocker.

AlexPoulsen, 2016-11-22:
@kristovatlas Unless the attacker is able to hack into a system process. Even then, I don't think much of macOS is actually running. One would have to test the CPU time of various processes, including kernel_task (which if anything is running, it is), and some other system processes to see what is running and what isn't. Compare the time before and after hibernation with a quick test (login as soon as it boots), and with a slow test (wait a while keeping the screen on) to check whether on not certain things are increasing. Then you know what to hack. And if nothing is increasing much in the long test, then it is firmware that is running the unlock-the-disk screen and idk if a usb device could modify firmware. Opening the computer up to modify the firmware isn't something one could do easily or when the target is in the bathroom.