samyk/poisontap #5: Would unloading kernel modules on OS X be an effective mitigation?

khourybrazil, 2016-11-17:
There's various kexts that are responsible for USB ethernet/networking:
/System/Library/Extensions/AppleUSBEthernet.kext
/System/Library/Extensions/AppleUSBNetworking.kext
/System/Library/Extensions/AppleUSBEthernetHost.kext

neuhaus, 2016-11-17:
That would interfere with the operation of existing USB/networking devices, wouldn't it?

khourybrazil, 2016-11-17:
Undoubtedly. AppleUSBEthernetHost would keep you from using your phone as a USB hotspot for sure.

This might be something you can do when you're outside of your home or office if you use a USB network device there.

samyk, 2016-11-17:
This would work, but be extremely inconvenient. I personally use the Belkin dock at home to connect Ethernet for high speed networking and this would prevent that from working, and I do use Ethernet adapters quite often to connect to BeagleBone/RaspberryPis.

It's good to mention as a mitigation but won't work for a lot of users unfortunately. Perhaps a tool that keeps them unloaded, but then pops a dialog to the user when it's detected a new USB device (usbtracer, system_profiler SPUSBDataType, USB Prober) and if user agrees, hot loads the extension. Not a perfect solution still though as PoisonTap could be added when enabled.

khourybrazil, 2016-11-18:
Agreed, it would be a huge pain. I can't imagine it would be worthwhile for anyone but the most security conscious.