samyk/poisontap #6: Does this work on raspberry pi 2/3?

ZoeLe, 2016-11-17:
This is more of a question than an issue. Thanks! :)

samyk, 2016-11-17:
Not sure if the USB ports or other GPIOs can act as USB slaves or not. Would be great if someone can test!

ZoeLe, 2016-11-17:
I just found this, https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=15696 it appears that model A allows it and model B (the one I have) does not. :( I am going to see if I can find some kind of work around...

JarrettR, 2016-11-17:
In this case, support is a function of hardware - The model A connects the external USB port directly to the Broadcom chip (which supports USB OTG), while the model B or 2/3 use an external USB host chip to drive multiple ports, and does not support OTG.

No software workaround possible, sorry :8ball:

ZoeLe, 2016-11-17:
OK, thank you!

samyk, 2016-11-17:
You may be able to bit-bang USB in userland over the GPIOs (instead of depending on hardware USB support)

IrishGhost, 2016-11-17:
it only works on RP zero? right?

IrishGhost, 2016-11-17:
Who checked?

ZoeLe, 2016-11-18:
It should work on the Raspberry Pi Model A, but I haven't checked it yet.

WolfyLPDC, 2016-11-18:
There are not Raspberry Pi Zero available here, and the shipping costs are more expensive than buying a full Raspberry. So i hope to know what kind of Raspberry should i buy... ¿Any idea?

t3chguy, 2016-11-18:
Zero and A both have OTG functionality and both have worked for g_ether for me personally so there is no reason that they should not work for this. All other Pis are excluded natively due to their use of a USB Hub.

D3sry3R, 2016-11-18:
Does this work on Raspberry Pi 3?

jreziga, 2016-11-18:
@D3sry3R No.

Z3r0byte, 2016-11-18:
Would it be possible to connect the Pi (non-zero) to a PC with an Ethernet cable using the port on the Pi and use a power bank to power it? I don't know much about all the chips and stuff in the RPi, but it sounds logic to me.

samyk, 2016-11-18:
That should work if the PC has Ethernet. You could even pull power from USB of the PC.

samyk, 2016-11-18:
Alternatively, another way to use any of the Raspberry Pis that have ethernet is to use Raspberry Pi -> ethernet cable -> ethernet-to-USB/Thunderbolt adapter, then plug it into the target machine via USB/Thunderbolt. The same exploit chain will work like normal, though you'll see some errors from pi_startup.sh and you'll need to adjust all instances of `usb0` to `eth0`.

Z3r0byte, 2016-11-18:
@samyk I changed all 'usb0' to 'eth0' an removed the USB setup part. However, I am getting the following errors when trying to execute the script:
```
root@Frambozentaart:/poisontap# ./pi_startup.sh
ifup: interface eth0 already configured
SIOCADDRT: File exists
Starting isc-dhcp-server (via systemctl): isc-dhcp-server.serviceJob for isc-dhcp-server.service failed. See 'systemctl status isc-dhcp-server.service' and 'journalctl -xn' for details.
 failed!
net.ipv4.ip_forward = 1
root@Frambozentaart:/poisontap# systemctl status isc-dhcp-server.service -l
● isc-dhcp-server.service - LSB: DHCP server
   Loaded: loaded (/etc/init.d/isc-dhcp-server)
   Active: failed (Result: exit-code) since Fri 2016-11-18 20:19:25 GMT; 8s ago
  Process: 2159 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)

Nov 18 20:19:23 Frambozentaart dhcpd[2166]:
Nov 18 20:19:23 Frambozentaart dhcpd[2166]: No subnet declaration for eth0 (192.168.2.1).
Nov 18 20:19:23 Frambozentaart dhcpd[2166]: ** Ignoring requests on eth0.  If this is not what
Nov 18 20:19:23 Frambozentaart dhcpd[2166]: you want, please write a subnet declaration
Nov 18 20:19:23 Frambozentaart dhcpd[2166]: in your dhcpd.conf file for the network segment
Nov 18 20:19:25 Frambozentaart isc-dhcp-server[2159]: Starting ISC DHCP server: dhcpdcheck syslog for diagnostics. ... failed!
Nov 18 20:19:25 Frambozentaart isc-dhcp-server[2159]: failed!
Nov 18 20:19:25 Frambozentaart systemd[1]: isc-dhcp-server.service: control process exited, code=exited status=1
Nov 18 20:19:25 Frambozentaart systemd[1]: Failed to start LSB: DHCP server.
Nov 18 20:19:25 Frambozentaart systemd[1]: Unit isc-dhcp-server.service entered failed state.
root@Frambozentaart:/poisontap#
```

It says I have to do something with dhcp.conf, however I can't find that file anywhere. Can you (or someone else) please help me?

pLesur, 2016-11-18:
Could it work on a RPi 1 B ? I feel like it'd be a waste of time trying instead of just getting a RPi zero, but maybe someone got it working ? Thanks.

Z3r0byte, 2016-11-18:
+pLesur Right now I am trying to get this to work on RPi 1 B, but as you can see in my post above, it doesn't work yet.

jreziga, 2016-11-18:
@Z3r0byte I've the same issue. Referenced twice here: https://github.com/samyk/poisontap/issues/16 and here: https://github.com/samyk/poisontap/issues/19

samyk, 2016-11-18:
@Z3r0byte Added dhcpd.conf in 838a0353c606cc815b26e290e8347a08e18312ba

samyk, 2016-11-18:
Additionally, added further installation instructions in 26dd7818a8f74daa81e156a106cde8fffdc19004

Z3r0byte, 2016-11-19:
Thanks! Its working now. :)

closed 2016-11-19 by samyk

joh1234, 2016-11-19:
https://goo.gl/JBljdR

ZoeLe, 2016-12-01:
I am getting this error when I run the pi_startup.sh script on my pi2:
<pre><code>
./pi_startup.sh: 8: cd: can't cd to /sys/kernel/config/eth0_gadget/
ln: failed to create symbolic link 'configs/c.1/acm.eth0': File exists
ln: failed to create symbolic link 'configs/c.1/ecm.eth0': File exists
ls: cannot access /sys/class/udc: No such file or directory
ifup: interface eth0 already configured
SIOCADDRT: File exists
./pi_startup.sh: 42: ./pi_startup.sh: /etc/init.d/isc-dhcp-server: not found
net.ipv4.ip_forward = 1
./pi_startup.sh: 46: ./pi_startup.sh: /usr/bin/screen: not found
./pi_startup.sh: 47: ./pi_startup.sh: /usr/bin/screen: not found
</code></pre>

Thanks!

samyk, 2016-12-01:
@ZoeLe running with sudo/root?

ZoeLe, 2016-12-01:
yes, I am running this as root.

samyk, 2016-12-01:
Oh, most of pi_startup.sh is meant for the Pi0. Pi2/3 need to change things to use eth0 instead of usb0. Also you're missing a number of packages that aren't installed -- see the install section in the readme.

ZoeLe, 2016-12-01:
Oh. Sorry I did not see the installation section :|

SpaceMan1228, 2017-01-24:
@samyk Hi, sorry to bring this topic up again. But basically, I followed your previous steps. Changed the installation, and the dhcp.conf file. And I did the thing, where when the rpi3 boots up, it runs the startup file for poisontap. But anyways, it doesn't seem to work. The usb ports don't work (my keyboard doesn't work when I plug it in, but it works on my laptop). Also the Ethernet wont work either. Usually the light of the rpi3 Ethernet port turns on when you plug in the Ethernet, to say, a router. But Ethernet doesn't work! Help is appreciated from anyone. Thanks!