samyk/poisontap #7: Mobile device support

madmod, 2016-11-17:
Could this be made to work on locked Android phones? I'm assuming they would need otg host enabled by the user first right? (I don't personally own an Android device but I may borrow one once I get my pi zero.) It would be interesting to see if common android background requests could be abused and how they are evaluated. I doubt there is anything happening in the background which renders in a web view except maybe those network proxy authentication pop-ups some hotels use. (Sorry don't know the term.) Are there common third party/vendor launchers which use a web view we could inject into? Are there other features accessible from the lock screen like Google assistant we could trick into loading a web resource? if any the browser prefetch pages you ask assistant to load before unlocking?

What about a Bluetooth network device that asks to pair with a target phone? Obviously this would require some user interaction to pair but in the right circumstance it could be convincing. (Spoof car name when the target enters a car for example.)

As far as I know the only networking that happens on iOS over lightning is hosted by the phone for tethering which might prevent this attack.

madmod, 2016-12-01:
I have done some basic testing with the iPhone 7 Plus using the USB 3.0 lightning adapter for the iPad Pro and am super happy to report that it actually connects as a network interface without even needing external power! It looks like requests are not being intercepted in Safari or Chrome however going to http://1.0.0.1 does show the posiontap page. I tested http://nfl.com and a few others that worked on my Mac in Chrome.

Also awesomely I can connect to SSH on the pi using Prompt which makes this really useful combo for a lot of projects! It's a very easy setup to have a fully portable linux system with a ton of GPIO and a nice touch screen! Some kind of web interface for poisontap to see the logs of previous runs on a connected mobile device would be a great addition. Also being able to arm/disarm the device for testing and change other parameters on the go like the backdoor target, add custom domains, and to show/hide the screen would be great.

I have not tested this while the device is locked yet. I need to figure out how to capture more detailed info on the requests going to poisontap.

Also how can I undo the network changes poisontap makes so that I can connect the pi to wifi again? My usb wifi adapter no longer works after installing poisontap and `ifdown usb0` does not fix it.

zeroiq69, 2017-12-29:
if you modify the the code and and get it to work can you send it to me I want to see if it'll work on something