samyk/poisontap #8: Windows support

aizelauna, 2016-11-17:
Does poisontap really work with Windows 7 or 19? 
AFAIK, CDC ECM is not supported so Windows should never see the device as an ethernet device.

strasharo, 2016-11-17:
At least on Windows 7 it's supported, but you have to point it to the driver manually like this:
https://github.com/ev3dev/ev3dev/wiki/Setting-Up-Windows-USB-Ethernet-Networking

Not sure if it's possible to make it automatically recognize it.

Plazmaz, 2016-11-17:
Last I checked, this has no support for Windows 19.

aizelauna, 2016-11-17:
So I think we agree to say that poisontap cannot hack a Windows PC as described in the README.

brew-ninja, 2016-11-19:
confirmed working on win10

mwwhited, 2016-11-22:
Manually installing a hack doesn't count as it working.

brew-ninja, 2016-11-22:
Windows 10 required no intervention. Auto installs.

aricel, 2016-11-23:
tested on win7 x64 sp1, win8.1 x64, win10 and not working: the pi zero is detected as "Other devices => RNDIS/Ethernet Gadget". any hints to make it auto install?
thanks

PocoCraft, 2016-11-23:
I've tested this on Windows 10 without any signs of the RNDIS device showing up in Device Manager.. I may need to reinstall Raspian to verify that this is not a problem with preexisting/conflicting software.

mwwhited, 2016-11-23:
The only way to get this to "auto install" is to manually exploit your machine by installing the driver then using the device.  (This requires the machine to be unlocked and the user to have elevated privileges.)

After the driver is installed then this "attack" will work.  But the exploit was the driver install not this device.

Eastonboy99, 2016-11-23:
Windows 10 works without a problem. I have my target machine on wifi and when the pi is plugged in, it connects to the Ethernet gadget and launches PoisonTap.

samyk, 2016-11-23:
I've also tested on a Windows 10 machine without requiring installing anything. Perhaps some versions come with the driver installed. @mwwhited sounds like you're more familiar with this area, do you want to provide some suggestions in the case of the systems that do not autoload? Do you know if all ethernet-to-USB dongles use this same driver or other drivers that may provide automatic access across a wider base of users? Perhaps there's another driver we can emulate on the RPi.

brew-ninja, 2016-11-23:
I just tried on a second machine - vanilla win10 pro install. Device auto installs and comes up - it did fail the first time I plugged it in (driver installed but device issue) I unplugged and plugged back in and it came up fine. 
Just wanted to validate that I hadn't previously installed driver manually on the first machine I tried it on.

Stephen2929, 2016-11-26:
I've tried several Win10 systems, and whilst the 'PoisinTap' USB Composite device installs, the CDC ECM nic driver does not

brew-ninja, 2016-11-26:
installs fine without interaction on my win 10 systems

[image]

this is the driver that is used.
[image]

samyk, 2016-11-27:
Thanks for sharing @brew-ninja, appreciate the help!

Stephen2929, 2016-11-27:
Thanks. Could you do me a favour and confirm what "Bus reported device description" and "Hardware Ids" shows for that device

brew-ninja, 2016-11-27:
Bus reported device description: RNDIS/Ethernet Gadget
Hardware Ids:
USB\VID_0525&PID_A4A2&REV_0404
USB\VID_0525&PID_A4A2

Blocking the device ID via local/group policy was the basis of this mitigation I suggested: https://github.com/samyk/poisontap/issues/26

Lucg, 2016-11-28:
On some machines it is installed and recognized as a serial port, thus not working. The only solution in such a case is to take the serial port, update manually the drivers to the Acer RNDIS/Ethernet Gadget and then it will work. It was not working at first for me (but it is not a Poisontap related issue, more of a raspberry installation issue)

mmourey, 2016-12-05:
I have tested poisontab in three different windows seven pro machines, in each case installing the Rdnis driver manually is required for the ethernet adapter to work. I have found this page and a Pdf
[HCC RNDIS Device Class Driver Windows Automatic Installation Guide v1.00.pdf](https://github.com/samyk/poisontap/files/631503/HCC.RNDIS.Device.Class.Driver.Windows.Automatic.Installation.Guide.v1.00.pdf)
 that talk about wich usb_class and subclass to advertize in order for automatic driver installation : 

https://social.msdn.microsoft.com/Forums/en-US/56cd7c13-0c83-4c58-9b20-a25423dd17fd/how-to-make-automatically-device-driver-installation-in-win-7win-8-when-using-rndis-in-windows?forum=winembplatdev

I have tried adding a bDeviceSubClass with a value of 0x04, with no luck but I'am quite sure we can find a way to make it work.

By the way thanks for the awesome work !

mmourey, 2016-12-05:
I did some more poking, and I found that the guys at ev3 (linux for Lego Mindstorm) have found a usb gadget config that appears to works : https://github.com/ev3dev/ev3-systemd/blob/ev3dev-jessie/scripts/ev3-usb.sh Will try it tomorrow and make a pull request if it's really working.

aricel, 2016-12-08:
I modified the script:
echo 0x04b3 > idVendor # 
echo 0x4010 > idProduct #
and the network device is installed without user interaction on a win7 x64 sp1 as 
"IBM USB Remote NDIS Network Device". anyone to verify on win8 / win10?

samyk, 2016-12-08:
Oh awesome! I don't have Windows available anymore but I will attempt to test on macOS 10.12.1. Someone able to test on Linux would be helpful too.

aricel, 2016-12-08:
kali install =>

$ dmesg
[196997.716921] rndis_host 3-3:1.0 usb0: register 'rndis_host' at usb-0000:00:14.0-3, RNDIS device, 

$ lsusb
Bus 003 Device 044: ID 04b3:4010 IBM Corp.

$ ifconfig
usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500

lsfxz, 2016-12-08:
@aricel 
Seems to work on win 8.1 64. Considering the issues I had on 10 were the same as on 8.1 I'd guess it should work on 10 as well.

aricel, 2016-12-09:
works on win10

mmourey, 2016-12-09:
Fiy on my rasbian when using the g_ether module, the configfs information in the startup.sh script is not taken into account and the usb gadget default to standard value, I have to load the libcomposite module in order for the configfs to be used.

AddaxSoft, 2017-01-04:
@mmourey - I managed to get this working on Windows 8 & 10 with auto installing the driver. Thanks for the hints you included in your previous posts :)

However, I failed to do the same for Windows 7. I'm not sure if it's a protection mechanism for the image I have. Did you manage to get it working on Win 7?

see attached img from Win 8 of the rpi zero without any additional driver (or manual installation)
[image]

samyk, 2017-01-04:
@AddaxSoft did you try the EV3 USB VID/PID in pi_startup.sh?

```sh
echo 0x04b3 > idVendor
echo 0x4010 > idProduct
```

chuspepe79, 2017-01-17:
Hello,
I have been testing with poisontap and I have made several clean installations but it still does not work. I followed the installation instructions step by step and nothing. In Windows 7 poisontap appears with RNDIS Ethernet Gadget without associated driver, I tried to install a driver but without success. In Windows 10 poisontap appears as a Serial USB device and I could not change the driver. I have already tried changing the ID Vendor and ID Product and without success as well. In Linux if I launch an lsusb if the device appears to me but I am not launched the capture process. I do not have any Macs to run the test.
Thank you very much for the help.

samyk, 2017-01-17:
You may want to try this version which has automatic OS detection and adjusts setup based on it:
https://github.com/wismna/HackPi

PrettyBoyPBM, 2017-03-18:
@chuspepe79 did it worked?

PrettyBoyPBM, 2017-03-18:
@aricel hi can i ask u a few questions via email or something?

zbozic, 2017-03-29:
@sammyk 
Can I reinstall HackPi over your original version which I already install on my raspberry pi zero? Or i need to deistall your version first? 
Thx.

HOD42, 2017-12-01:
I was having real problems with this (after initially having it working and then corrupting my SD card). It turned out that things were fine using a cable but I'd re-done my soldering of the USB-Stem for the PiZeroW and now it's no longer being detected by Windows. 

To clarify; the same PiZero is detected by Windows 10 as "unidentified device" when connected via the stem but is correctly identified as RNDIS/Ethernet when connected via a micro USB cable.

I may need to re-re-do my (dodgy) soldering ;-)

GavinDarkglider, 2018-03-13:
There is a better way to fix this issue, than using a the vid/pid option.... While this works, you might not want your end device to use those values.....

This is where the os descriptors come in

echo 1 > os_desc/use
echo 0xcd >os_desc/b_vendor_code
echo MSFT100 > os_desc/qw_sign
echo RNDIS   > functions/rndis.usb0/os_desc/interface.rndis/compatible_id
echo 5162001 > functions/rndis.usb0/os_desc/interface.rndis/sub_compatible_id
ln -s configs/c.{Configuration you linked functions/rndis.usb0 to} os_desc/

If you add this to the RNDIS config section, you can keep any VID/PID, and windows will always detect the driver. That being said, if someone knows how to get RNDIS to work on windows when part of a composite gadget, that would be really helpful.