samyk/poisontap #24: SyntaxError when running backend_server.js in Node

JonathanNathanson, 2016-11-19:
Hi, 

I'm getting a bunch of SyntaxErrors thrown by NodeJS when trying to run the backend_server.js on my server. I've tried the repo I pulled yesterday, and then re-cloned the latest version via Git just to make double-sure..

`|- /home/jono 
jono@ubuntu:~$ 
git clone https://github.com/samyk/poisontap.git
Cloning into 'poisontap'...
remote: Counting objects: 370, done.
remote: Compressing objects: 100% (83/83), done.
remote: Total 370 (delta 295), reused 360 (delta 285), pack-reused 0
Receiving objects: 100% (370/370), 4.34 MiB | 2.91 MiB/s, done.
Resolving deltas: 100% (295/295), done.
Checking connectivity... done.
`

Trying to launch throws this error:

`nodejs backend_server.js `

`
/home/jono/poisontap/backend_server.js:9
var server = http.createServer((request, response) => {
                                                    ^`

`SyntaxError: Unexpected token >
    at Module._compile (module.js:439:25)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:902:3`

It's not displaying particularly well in this but I think you get the gist. Anyone else having this issue?

1m0r74l17y, 2016-11-19:
i am having the same error

jameshhx, 2016-11-19:
You'll need to install a version of node that supports ES6 features. Try the v6 binaries from https://nodejs.org/en/download/ for ARMv6.

joh1234, 2016-11-19:
Update README
https://goo.gl/JBljdR

JonathanNathanson, 2016-11-19:
@joh1234 link is malware, go away.

JonathanNathanson, 2016-11-19:
@jameshhx thanks, worth noting then that the standard Ubuntu 16.04 package doesn't support ES6 features.

jameshhx, 2016-11-19:
No worries. You'll also need to install the websockets module. Only node dependency as far as I can see.

`npm i websockets`

Doc1992, 2016-11-20:
I'm getting the same error as well when trying to run this file with nodejs.

I'm on raspberry pi 2, connected to victim PC (windows 10 laptop) with Ethernet cable (no Ethernet to USB dongle). The victim pc successfully gets an IP of 1.0.0.10 but shows no sign of PoisonTap working other than slow browsing or pages not loading after long loads.

Am I supposed to run anything else on the Pi? How can I check to see if it's working or not?

JonathanNathanson, 2016-11-20:
@Doc1992 PoisonTap, it it's released form, is intended for use on a PiZero, connecting to target machines via USB, not Ethernet. 

Your question is not specific and you have not provided any details. Your question is very general and seemingly entirely unrelated to this particular issue thread.