samyk/poisontap #27: Having trouble getting things working on USB Armory

sporkmonger, 2016-11-20:
So first up, I did host the backend on Heroku, so that _might_ be part of the issue, but I'm reasonably confident I set up all the references to the backend correctly and the backend at least does appear to receive the request (though it hangs indefinitely).

As for the Armory, when I plug it in (this is all macOS Sierra), takes about 5-10 seconds and then it shows up in the network devices list on top. "Configure IPv4: DHCP", apparently gets an IP in the link local range (169.254.70.148 in my case). Does not respond to 1.0.0.1. Also can't SSH into the Armory anymore. Machine can still talk to the outside internet without issue.

I did all the setup on this machine, but before plugging it in, I removed it from the devices list and turned off connection sharing to the device so that it'd be starting from a blank slate without any of the setup steps I took to get things configured still in place.

/cc @abarisani in case you're interested.

sporkmonger, 2016-11-20:
[image]

samyk, 2016-11-20:
Looks like DHCPD issue. Are you using the provided dhcpd.conf and is dhcpd running? Is the machine IP set to 1.0.0.1?

samyk, 2016-11-20:
BTW, (unrelated) when you plugged it into OS X, did it require any manual accepting of the device or did it just come up in the Network prefs without question? Also what version of OS X?
Eg, did you see something like https://pbs.twimg.com/media/CxpVVnPUsAA5DJA.jpg:large

Another user experienced that screenshot but I've never seen it and want to see if it's perhaps OS X version related.

sporkmonger, 2016-11-20:
I used the provided dhcpd.conf file, yes, and confirmed that it had been copied over correctly. `/etc/network/interfaces` was configured for 1.0.0.1.

sporkmonger, 2016-11-20:
No manual accept required. It just popped up there at the top. macOS 10.12.1. I didn't get a notice like that, but it also wasn't the first time the machine had seen the Armory?

samyk, 2016-11-20:
I would tail /var/log/messages and any other logs dhcpd may be logging to, then see what it says when booting up (or if there's a foreground/verbose mode of dhcpd). You should be issued an IP in the 1.0.0.x range. Also, is it set to the right interface? Maybe dhcpd.conf/network interface needs to change when using USB Armory (vs usb0 on RPi0)

sporkmonger, 2016-11-20:
Since the normal ssh connection on the armory is on usb0 w/ static IP, I'd assume it's same as RPi0.

sporkmonger, 2016-11-20:
Hard to tail the logs since I can't connect to it anymore. I'll yank the microSD and see what's there though.

samyk, 2016-11-20:
Ah, okay, logs are next then.

Thanks on the other point. I'm also on 10.12.1, and have also tested on either 10.10 or 10.11 (I forget which), and have never seen that popup. Also used USB Armory and haven't seen it. Thanks for verifying.

sporkmonger, 2016-11-20:
I'll probably test on a fresh OS X machine soon.

samyk, 2016-11-20:
You may be able to run it as serial gadget and `screen /dev/tty.whatever 115200` in.

Great, would love to hear the results on a fresh OS X machine if you do test! Heading to bed but will check in on logs in the morning.

abarisani, 2016-11-20:
Not sure if this helps but for best results we pass the use_eem=0 to the g_ether module when bringing up the RNDIS device.

We did stumble upon a OSX bug where the RNDIS device, when presented with a different MAC address, would show up in network conf as "green" but not be actually up (requiring manual ifconfig up), so it is also important to keep the same host/device MAC pair, but I see poisontap is doing this already so it shouldn't be an issue.

In order to ensure that your OSX setup is correct might be helpful trying with our Debian image (which has static IP address 10.0.0.1) first, maybe also enable its DHCP server later (it's present and configured, just not started by default).

In general we never see issues with USB armory networking on OS X so this should work just fine.

sporkmonger, 2016-11-21:
@abarisani I used the Armory Debian image as my base. I followed the PoisonTap install instructions for DHCP though.

abarisani, 2016-11-21:
Then I am not sure what is going on but there are no hardware dependencies in the process, DHCP is hardware neutral. So whatever the problem is it shouldn't be specific to the USB armory but some interaction between the Debian image and the process you are following.