samyk/poisontap #28: ls: write error: Device or resource busy

JetseVerschuren, 2016-11-20:
When i try to run pi_startup.sh i get a error:
[image]

BenjaminHae, 2016-11-20:
If figure it's about the line `ls /sys/class/udc > UDC`. Before running the script the file `/sys/kernel/config/usb_gadget/poisontap/UDC` doesn't exist. Afterwards it does. I don't have a clue why the script can create it, but can't write to it.

samyk, 2016-11-20:
Did you follow the installation instructions in README.md?

JetseVerschuren, 2016-11-20:
Yes i followed the installation instructions at the bottom of readme.md

samyk, 2016-11-20:
So you're saying pi_startup.sh is already in /etc/rc.local per the instructions, but you're running it a second time as demonstrated in your screenshot? You should only run it once.

BenjaminHae, 2016-11-20:
I followed everything but the addition to `rc.local` so I could see possible errors when running it manually - as it doesn't work.

It seems to be the same error when the script is executed in `rc.local` because after booting`/sys/kernel/config/usb_gadget/poisontap/UDC` exists but is empty.

samyk, 2016-11-20:
May want to try following https://gist.github.com/jgamblin/459d7db3b4e8441cc210b35abb5d7181 or https://gist.github.com/gbaman/50b6cca61dd1c3f88f41 in case there is something I missed in the installation

JetseVerschuren, 2016-11-20:
It partially works on my ubuntu computer, it only runs when i type 1.0.0.1 in the browser and not all requests are made to the pi zero.
Also windows doesn't recognize the pi as a network adapter.

JetseVerschuren, 2016-11-20:
Why did you use a 128.0.0.0 subnet and not a 0.0.0.0 subnet?

samyk, 2016-11-20:
What version of Windows?

128.0.0.0 works on every OS I tested where 0.0.0.0 doesn't. I do want to update dhcpd.conf to use a short lease time, and have the subnet dynamically switch between the lower half and upper half of the IPv4 space but haven't had a chance to, where the usb0 interface will hold an IP in both halves.

JetseVerschuren, 2016-11-20:
I am using windows 10.

closed 2017-09-05 by JetseVerschuren

combatopera, 2019-08-24:
the problem seems to be the one-liner, splitting it worked for me:
```
name="$(ls /sys/class/udc)"
echo "$name" >UDC
```