samyk/poisontap #30: Running on RPi 2 without Ethernet to USB

Doc1992, 2016-11-20:
I'm using a Raspberry Pi 2 powered by the power socket and connected to the victim (Windows 10 Laptop). I can successfully run `./pi_startup.sh` with no errors after doing the following steps:

- Installation instructions from README without the USB stuff and replaced all instances of usb0 with eth0 in `/etc/network/interfaces`:
```
echo -e "\nauto eth0\nallow-hotplug eth0\niface eth0 inet static\n\taddress 1.0.0.1\n\tnetmask 0.0.0.0" >> /etc/network/interfaces
echo "/bin/sh /home/pi/poisontap/pi_startup.sh" >> /etc/rc.local
mkdir /home/pi/poisontap
chown -R pi /home/pi/poisontap
apt-get update && apt-get upgrade
apt-get -y install isc-dhcp-server dsniff screen nodejs
cp /home/pi/poisontap/dhcpd.conf /etc/dhcp/dhcpd.conf
```

- Replaced `pi_startup.sh` from @Z3r0byte [https://github.com/Z3r0byte/poisontap/blob/master/pi_startup.sh](url) 

- Connected to victim PC (has chrome open) with an Ethernet cable from RPi Ethernet port to victims Ethernet Port and rebooted the RPi. The victim PC successfully gets an IP of 1.0.0.10. Browsing is slower and spotty but apart from that nothing shows that PoisonTap is working. No cookies or such appear in `/home/pi/poisontap/` . Am I supposed to run anything else on the RPi?

When attempting to run `nodejs backend_server.js` I get the same error as #24 

```
/home/pi/poisontap/backend_server.js:9 var server = http.createServer((request, response) => {
SyntaxError: Unexpected token > at Module._compile (module.js:439:25) 
at Object.Module._extensions..js (module.js:474:10)
at Module.load (module.js:356:32)
at Function.Module._load (module.js:312:12)
at Function.Module.runMain (module.js:497:10)
at startup (node.js:119:16)
at node.js:902:3 
```

pmbenjamin, 2016-11-21:
The `Unexpected token` error message you're getting is basically Node complaining that it doesn't understand the new [Arrow Function](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Functions/Arrow_functions) syntax.

This is because you're most likely using an older version of NodeJS.

This can be resolved in 1 of 2 ways:
- _EASY (UPGRADE NODE):_ Install a newer version of NodeJS (not the default one from `apt-get` repositories). I would recommend installing [nvm](https://github.com/creationix/nvm) (Node Version Manager), then `nvm install 6.9.1` (or higher).
- _DIFFICULT (CHANGE FUNCTION SIGNATURE):_ If you have a requirement to use an older version of node, or if upgrading node is not an option, you will have to replace all instances of the new Arrow Function syntax (e.g. `(...) => {...}`) with the traditional Function syntax (e.g. `function(...) { ... }`). Of course `...` indicates some code that must not be changed in this process.

Doc1992, 2016-11-21:
PoisonTap runs `backend_server.js` automatically? How can I check if PoisonTap is working?

pmbenjamin, 2016-11-21:
I'm not sure where or how PoisonTap runs `backend_server.js` automatically.
I see that PoisonTap runs `pi_poisontap.js` automatically from [here](https://github.com/samyk/poisontap/blob/master/pi_startup.sh#L47), however.

JonathanNathanson, 2016-11-21:
@Doc1992 Your problem has been answered, twice. backend_server.js does not run automatically, you need to set it up as a running nodejs app on an internet-accessible server, yourself, using a version of Node that supports ES6...

Z3r0byte, 2016-11-21:
You should execute pi_startup.sh, which starts the server, however, you've already placed a command in rc.local, so it gets executed automatically when you boot the RPi. ;)

mpereira13, 2016-11-30:
@Doc1992 check if your isc-dhcp-server is up, if is down check your isc-dhcp-server configs. I have some issues with this, and now it's work nice with my RPI2 and RPI3. ;)

Doc1992, 2016-12-03:
I got it to work after fixing the node version (NOT nodejs) on the pi, although it wouldn't work when started on boot from rc.local, had to start it up manually. And had some weird issues, on some runs it would work, some it didn't.

Working with a RPi zero now and gonna see how it goes.

closed 2016-12-03 by Doc1992