samyk/poisontap #34: log.php not included - by design?

JonathanNathanson, 2016-11-21:
Hi,

I think I know the answer to this.. but the git repo doesn't include log.php. I'm guessing this is because it's a simple script that I should write myself just to log all input from the target_backdoor.js?

Thanks,

samyk, 2016-11-21:
Yes, it was just a placeholder to demonstrate you could throw anything in there (this is specifically to backdoor HTTP JS objects like Google CDN jQuery). Technically you don't even need a script in its place if you can tail your access log but happy to accept a pull request if people would like to add to the backdoor :) Additionally, the code in backdoor.html would make a good backdoor (though the `content` div would need to be created via `document.createElement()` and any HTML stripped as this would become purely a JS backdoor)

zenware, 2016-12-16:
This totally reminded me that I wrote a [Web Socket botnet POC](https://github.com/zenware/botnetjs) when I first discovered Web Sockets. I set up a `botnet.js` file which connected to the WS server to communicate response and receive commands and a `/control` endpoint which was effectively just a textbox and a submit button that would then `eval(theInput);`against all the machines which had loaded that file. And then, of course, the goal was to MITM an HTTP connection in some way or get a connection to an HTTPS server in your control to inject the botnet file. This now seems super applicable to what you're doing here...