samyk/poisontap #37: Enhancement: Break out each exploit code into a separate module

pmbenjamin, 2016-11-21:
@samyk how do you feel about breaking out the exploits into independent modules (e.g. network hijacking, cookie siphoning, remote web backdoors, remote router backdoor) that can distributed through NPM and potentially be re-used in other projects?

samyk, 2016-11-21:
Sounds cool. They're separated out into functions, although the cookie siphoning + cache poisoning are joined for efficiency (cookie siphoned upon HTTP request, cache poisoned via the response)

pmbenjamin, 2016-11-21:
Are those functions confined to [target_injected_xhtmljs.html](target_injected_xhtmljs.html)?

samyk, 2016-11-21:
Well, all of the attacks require the injection of target_injected_xhtmljs.html + the running Node server (pi_poisontap.js) to deliver payloads and log cookies.