samyk/poisontap #38: hey,I have some questions about the poisontap，Please help me，very very thanks...

54zhangxun, 2016-11-22:
@samyk @jreziga 
I'm on raspberry pi 3B, connected to PC(win 10), and use a usb-cable to connect PC. It is right?
Q1:
   I'm looking this guid [part2](https://github.com/jreziga/poisontap/blob/068c2836deb61b150946acf3395ccf11eb3b3ad1/README.md)
And I tried to run pi_startup.sh, launch throws this error:
root@raspberrypi:/home/pi/poisontap# chmod +x pi_
pi_poisontap.js  pi_startup.sh    
root@raspberrypi:/home/pi/poisontap# chmod +x pi_startup.sh 
root@raspberrypi:/home/pi/poisontap# ./pi_startup.sh 
./pi_startup.sh: 8: cd: can't cd to /sys/kernel/config/usb_gadget/
ln: failed to create symbolic link ‘configs/c.1/acm.usb0’: File exists
ln: failed to create symbolic link ‘configs/c.1/ecm.usb0’: File exists
ls: cannot access /sys/class/udc: No such file or directory
Cannot find device "usb0"
Failed to bring up usb0.
usb0: ERROR while getting interface flags: No such device
SIOCADDRT: No such device
[....] Starting isc-dhcp-server (via systemctl): isc-dhcp-server.serviceJob for isc-dhcp-server.service failed. See 'systemctl status isc-dhcp-server.service' and 'journalctl -xn' for details.
 failed!

Q2:
root@raspberrypi:/home/pi/poisontap# nodejs backend_server.js 

/home/pi/poisontap/backend_server.js:9
var server = http.createServer((request, response) => {
                                                    ^
SyntaxError: Unexpected token >
    at Module._compile (module.js:439:25)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3

root@raspberrypi:/home/pi/poisontap# node backend_server.js 
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE :::1337
    at Object.exports._errnoException (util.js:1026:11)
    at exports._exceptionWithHostPort (util.js:1049:20)
    at Server._listen2 (net.js:1257:14)
    at listen (net.js:1293:10)
    at Server.listen (net.js:1389:5)
    at Object.<anonymous> (/home/pi/poisontap/backend_server.js:50:8)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
root@raspberrypi:/home/pi/poisontap# nodejs backend_server.js 

What should I do???

ax-el, 2016-11-22:
If you want to use anything other than a lan turtle, usb armory or pi ZERO (and not 1 2 or 3), then you'll need a usb to Ethernet adapter. since you use a pi 3 you can't just connect it to your PC via USB.

54zhangxun, 2016-11-25:
@ax-el  Thank you for your answer.
You mean the pi 3 can not be used to connect PC?
Or Using pi 3 and a usb to Ethernet adapter can connect PC?