samyk/poisontap #41: Installation Smooth & Great But... Not Working...

bibi1959, 2016-11-23:
I run through the installation following all (i hope) aspects including the config.txt and cmdline.xt By booting up and connecting to my Mac - the eth card is there but any site/browser and program using inet connection just doesn't open.  As soon as i disconnect everything is back to normal. Sorry for the silly question... but does anybody knows what i did wrong? Please get back if you need more info. Bruno
[image]
[image]

belialboy, 2016-11-27:
Did you ever fix this. My PiZ seems to be in a similar state as yours. I'm seeing the same "PoisonTap is listening" lines in the log, but when it's plugged into my Win10 machine, I see tumbleweeds. Nothing. Nada.

Lucg, 2016-11-27:
Be sure that your /etc/rc.local was empty before trying to start PoisonTap. With a traditional rasp installation, the /et/rc.local has an exit 0 before the line that we have appended

belialboy, 2016-11-28:
Yep, I saw that and fixed it... actually I worked out the noob mistake I'd made. With the jumble of wires on my desktop I'd inadvertently plugged by PC into the PWR USB slot rather than the USB labelled one on the PiZero. Well that's 6 hours I won't get back!

PR #51 (2016-11-30): remove premature "exit 0" from rc.local
Fixes #41
commit f598a38f (2016-11-30): remove premature "exit 0" from rc.local

Fixes #41
closed 2016-11-30 by samyk