samyk/poisontap #44: Poisontap on Teensy 3.6 w/ SD card

jayPhantomic, 2016-11-24:
Allo sir. Would you consider this feasible? It has native eth support, can support spi flash, and can carry a micro SD. There is also whispers of this being able to interface sdram via DMA. I haven't personally confirmed that but will running tests soon. Thanks for your time on this project. Great work so far.

TheOncomingStorm, 2016-12-01:
Teensy/Arduino doesn't really have all the capabilities that a RasPi does. For example, the RasPi has a full OS, etc. but the Arduino would only be able to run one program. Yes, you could perhaps make that possible if you did a lot of development(I don't know much about the 3.6) but otherwise it's far easier to find a RasPi Zero or the like and use it.

samyk, 2016-12-01:
@TheOncomingStorm is correct, it is technically possible, but most everything except the client side Javascript code would likely need to be rewritten. Additionally, one of the critical pieces (Ethernet gadget over USB) is currently **not** implemented in the Teensy core, although that might be a project worth working on in the future as I originally wanted to use Teensy for this and similar attacks.