samyk/poisontap #48: Mac OS X mitigation

wkandek, 2016-11-28:
On my macOS Sierra the automatic network configuration only works when running as a "administrator" user. When running as a "standard" user a pop-up requests manual network configuration. This seems like an effective mitigation for the small additional effort to run as "standard".

samyk, 2016-11-28:
Oh, very interesting! Someone else ran into that popup and I could not reproduce for the life of me across various machines, as almost all machines are an administrator user by default, even though I require "unlocking" for all System Preferences, I still would not experience the popup.

Thanks for sharing this method of mitigation!

tranquillechat, 2016-11-29:
Hi,
I can confirm, on simple user osx, a popup ask for administrator login/pwd, tested on 10.10.5.
By default, the os locked the preference panel on user mode. You can change the wallpaper but not network or share for example.

samyk, 2016-11-29:
@tranquillechat What made you configure it in that way? I assume this means you have at least two accounts, the normal account along with a separate account with admin privileges, correct? I assume that wasn't the default configuration and that you manually added multiple accounts?

wkandek, 2016-11-29:
for me it was a "best practice" effort. Mac OS X makes it not too
uncomfortable to work this way, basically one has to type in the username
of an administrator plus the password, so that seemed like a small price to
pay for an (non-quantifiable) amount of additional security...

tranquillechat, 2016-11-30:
@samyk, effectively, for a use "at home", by default mac and windows create an administrator account. But, i'm in enterprise and alls macs/windows have only a simple user account. 
So, it's blocked for mac. Plus, Quote wkandek "basically one has to type in the username of an administrator plus the password" is exact for me too.
But on windows, i got no problem to connect the PT with a simple user account (ad or not) :)

samyk, 2016-11-30:
Ahh, I see. All of my macOS machines (and friends machines) are home/office (non-enterprise) use, so makes sense why I've never experienced that. Thank you for the info!