samyk/poisontap #59: Mac: Works Once Only ?

Featherfx, 2016-12-04:
I got it to work on my macbook air
then 
I removed the RNDIS/Ethernet Gadget thing that gets installed under the Network manager in osx
I deleted the cache from safari to remove the backdoors.

But when I replug in my raspberry pi zero, its doesnt install the RNDIS/Ethernet Gadget again nor the backdoors? 

What i'm wondering is: is there some type of file that is created that tells the pi the machines that have been compromised already, so it doesnt re-do its thing? 

Note- it installs on others computers fine, but not the same one. hmmmmm

MarkMallett, 2016-12-12:
I have this exact same problem

Featherfx, 2016-12-12:
Actually I found a workaround, all you have to do is go to **Network**, click that little plus sign at the bottom left corner. Under Interface: drop down list, select **WIFI** and press create or **RNDIS/..** if you have that, I didn't have the RNDIS after I removed it, so I choose WIFI and just replugged in the pi zero and it worked again.

MarkMallett, 2016-12-12:
> Reply: samyk/poisontap <reply@reply.github.com>(mailto:reply@reply.github.com)

Either I'm doing it wrong or that didn't work for me.

--  
Mark Mallett

zenware, 2016-12-16:
I have a MBP, I'm in progress setting up the tool right now and after I've done it I'll test this out and see what I can come up with

zenware, 2016-12-16:
So I had a similar issue, what happened to me was that I had originally configured internet sharing as part of a headless setup process to be able to access the internet from the pi for updates. Before I disabled internet sharing and removed ` modules-load=dwc2,g_ether` from the end of my `/boot/cmdline.txt` file it wasn't operating correctly. At least I think that's what was going on, because before I disabled that it would still show the orange dot and 'Self-assigned IP', and now it's showing green and connected.
Though I did also try removing it to test it appearing without intervention, and while I was removing it from the list I was given the option "Would you like RNDIS/Ethernet Gadget to return next time it is connected to your Mac?" and I chose yes.
After that it took about 30 or 40 seconds to appear in the menu and be functional.