samyk/poisontap #63: Poisontap for beginners

mtlca, 2016-12-05:
hi guys!

I work in the computer department of a public school and we want to educate the community about computer security. We want to introduce them to this product. We buy a raspberry pi 3 in order to program the poisontap. But it seems to be more complicated than expected. Is there a detail procedure  from the beginning?  Or there are files to download that would make the task easier? Just upload some files on the raspberry. Iam new in programming. Thanks for helping the community. 

Thank you guys!!

crazy54, 2016-12-06:
I think this is a more broad question as you will need linux command line skills in order to complete the installation. The application is written for Node.JS, so you have to have to understand web servers and Node.  

Start here on your own and move on if you feel like this is something you should bring into the class: http://nodeguide.com/beginner.html

Maybe it would be better to just discuss this in class - Can you imagine you training 40 kids and then giving them this kind of tool?  Uh, does not seem like the smart way to go but that is just me...

FluxIX, 2016-12-06:
Security is a field where you look for gaps or holes in systems; you need to teach about the applicable systems (OS, web server, network communication and protocols, etc) before you can effectively use such a tool as this.

mtlca, 2016-12-07:
Thanks for the advice guys, but the issue here is that i want to do it. Lets begin: 
I launch this ans the node.js on my raspberry:
# Instructions adjusted from https://gist.github.com/gbaman/50b6cca61dd1c3f88f41
sudo bash

# If Raspbian BEFORE 2016-05-10, then run next line:
BRANCH=next rpi-update

echo -e "\nauto usb0\nallow-hotplug usb0\niface usb0 inet static\n\taddress 1.0.0.1\n\tnetmask 0.0.0.0" >> /etc/network/interfaces
echo "dtoverlay=dwc2" >> /boot/config.txt
echo -e "dwc2\ng_ether" >> /etc/modules
echo "/bin/sh /home/pi/poisontap/pi_startup.sh" >> /etc/rc.local
mkdir /home/pi/poisontap
chown -R pi /home/pi/poisontap
apt-get update && apt-get upgrade
apt-get -y install isc-dhcp-server dsniff screen nodejs

Whats the next step please!

mtlca, 2016-12-07:
Hi guys! 

I thin we can do it step by step:
1. I launch this on in my radpberry command line. 
-----------------------------------
# Instructions adjusted from https://gist.github.com/gbaman/50b6cca61dd1c3f88f41
sudo bash

# If Raspbian BEFORE 2016-05-10, then run next line:
BRANCH=next rpi-update

echo -e "\nauto usb0\nallow-hotplug usb0\niface usb0 inet static\n\taddress 1.0.0.1\n\tnetmask 0.0.0.0" >> /etc/network/interfaces
echo "dtoverlay=dwc2" >> /boot/config.txt
echo -e "dwc2\ng_ether" >> /etc/modules
echo "/bin/sh /home/pi/poisontap/pi_startup.sh" >> /etc/rc.local
mkdir /home/pi/poisontap
chown -R pi /home/pi/poisontap
apt-get update && apt-get upgrade
apt-get -y install isc-dhcp-server dsniff screen nodejs 
-----------------------
Whats next? Thanks for helping me. I can do it

mtlca, 2016-12-07:
Thanks guyz for the advices, but can i have a step by step tutorial like those on youtube. I can do it: 

1. I Execute those lines sucessfully in linux shell:

________________________________________________________

sudo bash
echo -e "\nauto usb0\nallow-hotplug usb0\niface usb0 inet static\n\taddress 1.0.0.1\n\tnetmask 0.0.0.0" >> /etc/network/interfaces
echo "dtoverlay=dwc2" >> /boot/config.txt
echo -e "dwc2\ng_ether" >> /etc/modules
echo "/bin/sh /home/pi/poisontap/pi_startup.sh" >> /etc/rc.local
mkdir /home/pi/poisontap
chown -R pi /home/pi/poisontap
apt-get update && apt-get upgrade
apt-get -y install isc-dhcp-server dsniff screen nodejs
_________________________________________________________

What is the next step? 

Thank you!

MrARM, 2016-12-07:
This will not run with the Raspberry Pi 3 hardware, as the raspberry pi zero has a special usb controller that supports usb hosts.

Check your local micro center for the Zero. got mine for $5.44 with tax.

mtlca, 2016-12-07:
Thank you very much MR ARM. Would that work with that? https://inversepath.com/usbarmory

zenware, 2016-12-16:
@mtlca Yes as the readme indicates it will work with the usbarmory, the lanturtle, the pi zero, and as @MrARM mentioned, anything which supports host mode
I'm currently working on a write-up as I install and configure everything from a coffee shop right now.