samyk/poisontap #66: PoisonTap not working on Win10 (Pi Zero)

DavidCatalan, 2016-12-07:
I have installed PT on my Zero, installed Node and checked that it is v6.9. It is working well on my arch machine though I've not checked if the backdoors are working properly, cookie siphoning is executed.

However on my Win 10 machine is not working, ipconfig shows that no IP like 1.0.0.10 has been configured.

mtlca, 2016-12-07:
Hi i would like to know how do you install it? 

Execute those lines in linux shell:

sudo bash
echo -e "\nauto usb0\nallow-hotplug usb0\niface usb0 inet static\n\taddress 1.0.0.1\n\tnetmask 0.0.0.0" >> /etc/network/interfaces
echo "dtoverlay=dwc2" >> /boot/config.txt
echo -e "dwc2\ng_ether" >> /etc/modules
echo "/bin/sh /home/pi/poisontap/pi_startup.sh" >> /etc/rc.local
mkdir /home/pi/poisontap
chown -R pi /home/pi/poisontap
apt-get update && apt-get upgrade
apt-get -y install isc-dhcp-server dsniff screen nodejs

what is the next step?

DavidCatalan, 2016-12-07:
Following the steps of the guide worked for me, don't forget to put all the files in their place, it is also at the guide. Check that node -v tells you that you are running node v6 and if not uninstall it and dowload the updated node sources from its original page.

mtlca, 2016-12-07:
Ok well thank you! Which guide you follow, where is the guide?