samyk/poisontap #67: Poisontap: Step by StepYoutube tutorial or procedure

mtlca, 2016-12-07:
Hi everyone, 

Where can i find a youtube tutorial or a step by step procedure on how to do a complete installation of poisontap?
Where can I find a detailed procture on how to program poisontap? I have to make a presentation on it in class. 

I sucesfully installed this lines in linux command shell:
_______________________________________________________________________ 
# Instructions adjusted from https://gist.github.com/gbaman/50b6cca61dd1c3f88f41
sudo bash

# If Raspbian BEFORE 2016-05-10, then run next line:
BRANCH=next rpi-update

echo -e "\nauto usb0\nallow-hotplug usb0\niface usb0 inet static\n\taddress 1.0.0.1\n\tnetmask 0.0.0.0" >> /etc/network/interfaces
echo "dtoverlay=dwc2" >> /boot/config.txt
echo -e "dwc2\ng_ether" >> /etc/modules
echo "/bin/sh /home/pi/poisontap/pi_startup.sh" >> /etc/rc.local
mkdir /home/pi/poisontap
chown -R pi /home/pi/poisontap
apt-get update && apt-get upgrade
apt-get -y install isc-dhcp-server dsniff screen nodejs
_____________________________________________________________________________-

What's next? 

Thank you for helping me!

Tibo-le-canard, 2016-12-12:
read the source code

Nexxic, 2016-12-20:
https://www.youtube.com/watch?v=dYDa_VDbjPY