samyk/poisontap #71: Stops all internet on the target computer

LoomyGloomy, 2016-12-11:
When plugged in it can successfully automatically install itself, but it kills the internet on the machine its plugged into. No animation displays, no cookies collected, the browser just reads unable to connect. I followed the installation guide to the letter and have node js current release. Any thoughts?

wdibi, 2017-01-05:
+1 I am having this same issue

samyk, 2017-01-05:
PoisonTap is supposed to take over the Internet for the computer, however I'm not sure why you're not seeing the animation. Perhaps an iptables rule is needed to forward all HTTP traffic to the PoisonTap IP (1.0.0.1). You could try ssh'ing to pi@1.0.0.1 and running:
```sh
echo "1" > /proc/sys/net/ipv4/ip_forward
iptables -t nat -A PREROUTING -p tcp --dport 80 -j DNAT --to-destination 1.0.0.1:80
iptables -t nat -A POSTROUTING -j MASQUERADE
```

nlacj, 2017-01-07:
Same issue here. Using Pi Zero with clean lite install + PoisonTap instructions.
I get the same result on Win10 or OSX.
Checked all the required settings and suspect iptables/routing related issue.

DHCP assigns an IP. (e.g. 1.0.0.15) No internet traffic once connected. 
I can manually open poisontap if I try an browse http://1.0.0.1.  But no internet once connected.

If I ssh onto the zero via 1.0.0.1, I can't ping any ip on the local network including the host I am coming from 1.0.0.15. Here is some output. I will continue to look at it when I get a few moments tomorrow.

root@poisontap:/etc# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         0.0.0.0         0.0.0.0         U     0      0        0 usb0
169.254.0.0     0.0.0.0         255.255.0.0     U     202    0        0 usb0

root@poisontap:/etc# ping 1.0.0.15
PING 1.0.0.15 (1.0.0.15) 56(84) bytes of data.
^C
--- 1.0.0.15 ping statistics ---
16 packets transmitted, 0 received, 100% packet loss, time 15000ms

root@poisontap:/etc# traceroute 1.0.0.15
traceroute to 1.0.0.15 (1.0.0.15), 30 hops max, 60 byte packets
 1  * * *
 2  * * *
 3  * * *
 4  * * *
 5  * * *
 6  *^C
root@poisontap:/etc#

wdibi, 2017-01-07:
I was able to solve this issue by restarting and following [these](https://github.com/jr-k/poisontap/tree/068c2836deb61b150946acf3395ccf11eb3b3ad1) instructions (scroll to bottom).

KALRONG, 2017-06-02:
I have a similar issue, in my case the animation works just fine and when I try to access nfl.com I get all the requests but, when I try to open nfl.com/PoisonTap for example, the websocket timesout (using a server ip instead of a domain, previously it was rejecting the connection I suppose due to it pointing to 1.0.0.1 in the resolution) and if I disconnect the PT I get a 404 error (I suppose due to not proper caching) but Im more interested in the issue while having the PT connected.

As @nlacj pointed out it seems like the problem is that the computer points to the zero but the zero doesn't actually have internet connection (well, Im using a zero W and if I leave the default route the zero doesn't have internet even when connected to the wireless) so when the computer tries to reach the internet, to reach the websocket for example, the connection gets stuck in the zero without a gateway.

Not sure if all my babbling will help but Im interested in fixing this issue.

himuura, 2017-11-16:
So, any solution to this issue? As soon as i connect the rasp, my internet connection goes out. Indeed it syphons cookies, but with an internet connection would be more realistic. I noticed that i only have the route on the script, the 0.0.0.0/0 gw 0.0.0.0. Tried to change the gateway, the OS broke. Anyone has an idea?