samyk/poisontap #74: Anti Virus

DavidPoe, 2016-12-13:
Hi 

it seems that the javascript backdoor used by poisontap is already recognized by some antivirus programs. (Symantec Endpoint Protection in my case). What would be the best way to disable the backdoor mechanism and just use the cookie siphoning functionality?

From a quick look at the code it looks like it is enough to clear out backdoor.html so no backdoor data would be sent to the client.

samyk, 2016-12-13:
Screenshot/link? I suspect they only recently added PoisonTap's code.

DavidPoe, 2016-12-13:
A Copy from the AV logfile:

Filename,Risk,Action,Risk Type,Original Location,Computer,User,Status,Current Location,Primary Action,Secondary Action,Logged By,Action Description,Date and Time
"6189D041BCED507C92DDE720EC2C3C64363A0D25","Backdoor.Trojan","Cleaned by deletion","Virus","C:\Users\XXXXXXXX\AppData\Local\Mozilla\Firefox\Profiles\tf51ux40.default\cache2\entries\","DEVXXXXXXX","XXXXXXXX","Deleted","Deleted","Clean security risk","Quarantine","Auto-Protect scan","The file was deleted successfully.","12/13/2016 6:37:24 PM"
"8AE7E31E9F26A45CE04E54993197D5D5B882B842","Backdoor.Trojan","Cleaned by deletion","Virus","C:\Users\XXXXXXXX\AppData\Local\Mozilla\Firefox\Profiles\tf51ux40.default\cache2\entries\","DEVXXXXXXX","XXXXXXXX","Deleted","Deleted","Clean security risk","Quarantine","Auto-Protect scan","The file was deleted successfully.","12/13/2016 6:37:30 PM"

Symantec Endpoint version 12.1.4013.4013
Definitions last Updated: 12th Dev 2016

fEDUntu, 2016-12-14:
Also there:
_target_injected_xhtmljs.html_ detected by Kaspersky Endpoint Security 10 Version: 10.2.5.3201 (mr3)
[image]

exploitagency, 2016-12-14:
I don't know if this works for html files but it does for exe, apk, etc.

www.nodistribute.com

Checks for detected signatures on all the popular scanners.

xfox64x, 2016-12-19:
There is neither a legitimate reason to point development efforts towards evading personal security products nor a reason why you cannot figure out a well-deserved answer to your own question through doing actual work. If one were conducting a legal and ethical penetration test, getting stopped by a security product would signal a failure in penetration efforts; go do better. Doing better, and conducting the research that warrants you an answer to this question, as already hinted at by exploitagency, means understanding what the PoisonTap scripts do, altering their behavior, and not getting your new (if any) payloads caught. Though, being that the capability these scripts make up can only serve two purposes (i.e. ethical and lawful penetration for educational/lawful means or malicious activities with the intent to violate privacy laws), it's reeeeeeeeeeeaaal difficult to assume that you have sufficient skill and/or good intentions for whatever you have planned. If you're running a pen-test and you really need to make that argument, "oh, yeah, we don't need the backdoor capability/we can pack/obfuscate the backdoor capability to avoid detection and steal all your stuff", you can just say that and your clients will probably believe you - it's not really a stretch to think that's in the realm of possibility, given understanding and skill. If you're trying to mess with your friends, scam some rando's, or commit other borderline criminal activity, you probably shouldn't be tipping your hand or asking for advice on how to accomplish that or help others accomplish that (and you should probably be able to figure it out, on your own).

25077667, 2016-12-20:
What's this?

exploitagency, 2016-12-20:
Just agreeing to the comment xfox64x posted.  Really no need to have a conversation on a Github issue, but Samy is the man and a friendly guy so it seemed acceptable to an extent.  I apologize if I offended any die hard Git'ers out there.  Even so, Merry GitMas anyways!  ( https://youtu.be/h6LlrQJS1Vc )