samyk/poisontap #75: Unable to SSH to Raspberrypi.local

MrSprawl, 2016-12-14:
I've tested this on 3 different micro SD cards, Tried both full and lite raspbian.  But following the guide of altering the config.txt and cmdline.txt after flashing (per the modular section) does not seem to work. I'm unable to get into raspberrypi.local.  Even after it connects and successfully grabs an IP address, I cannot SSH into that.  

I've noticed as of Nov of this year (2016) that SSH is disabled by default. I popped the SD cards, every one of them for both versions, into my Pi2 and enabled SSH just in case that was the issue. Am I missing something? I've tried this on Windows, Mac and Linux but I cannot ssh into the device when I plug it in. (I'm plugging it into the USB port, I've tried both)

ccamejo1, 2016-12-14:
Make sure you install the Bonjour Print Service for Windows on the PC.  This will allow you to connect to the pi using raspberrypi.local  You can find it here https://support.apple.com/kb/DL999?locale=en_US

C²

fEDUntu, 2016-12-14:
As @ccamejo1  said install the Bonjour Print Service for Windows (otherwise install iTunes, it works with me).
Follow this guide for more help:
https://www.thepolyglotdeveloper.com/2016/06/connect-raspberry-pi-zero-usb-cable-ssh/

MrSprawl, 2016-12-15:
Followed the instructions to the letter, even bought a new microsd card today and followed them. It starts up as a router and even gets an IP address yet I cannot for the life of me SSH into these devices on windows, mac or linux.  raspberrypi.local cannot be resolved and the IP address provided by the device is denied. I don't think it's the cable due to it working other than this SSH issue.

fEDUntu, 2016-12-15:
Awesome. Good hacking! :)

totalrisp, 2017-05-05:
If I follow that guide to the letter on a Debian machine, I only get a `enp0s20f0u2` device in `iwconfig`, that's it. Things like `arp -a` and avahi finder show nothing.

rahulmodi8812, 2017-07-24:
The SSH connectivity is by default DISABLED in latest raspberry pi. The RPi which are released in/after November 2016.

The steps are: 1) Write raspbian image to SD card. 2) create a ssh file without extension. 3) copy that ssh file to SD card where your image is written. 4) Finally wou will be able to SSH your raspberry pi.

For more details watch this video : https://www.youtube.com/watch?v=zKT6bTESlLs

sam0jones0, 2017-08-02:
> rahulmodi8812

You didn't read the post did you?

I am also having this issue. Thankfully I have a rp3 to try the hack suggested by MrSprawl

Addd12, 2021-04-21:
i managed to connect my pi through ssh before. then i went to another place and followed the same procedures to connect it again through the new network... so the ssh is enabled, i create ssh file and paste it to the ssd card and also changed the wifi credentials according to the new network in the wpa_supplicant.conf and it still doesn't work