samyk/poisontap #79: Unknown Failure

SpicyPiePete, 2016-12-17:
I had a few issues figuring out which of the slightly varied methods to set this up actually work. Currently, I have my server working using this to fix the node issues some of us all seem to be having:
# wget -O - https://raw.githubusercontent.con/sdesalas/node-pi-zero/master/install-node-v7.2.1.sh | bash
I do have No-IP configured, and I have my router setup to reserve the local IP and port forwarded 1337 to it. I've verified this is supposedly working with canyouseeme. My server is also showing that it's listening on that port, just like the picture this person took: https://github.com/samyk/poisontap/issues/76

Now the problem i'm having is with the device. Something is happening because the machines I've plugged it into are showing in the log,  but they are only showing the log that poisontap is listening. Nothing else happens. Also, when I watch the machine being exploited, I'm not seeing a new device or connection. Everything just works as normal. I'm also not noticing anything happening on the server side as this is going on. The only thing that happens is the random log entry. A couple notes: I have not configured this device to work through No-IP, but I have made sure to use the correct IP for the server that is. I also used the same fix that I used for the server, just to ensure node was the latest version. Everything else is a normal install.

Any guesses?

samyk, 2016-12-20:
Is dhcpd running? Does PoisonTap show up as an ethernet device on the machine?

SpicyPiePete, 2016-12-22:
I'm not exactly sure how to check if dhcpd is running, but PoisonTap is not showing up as an ethernet device on the machine. Going through the device manager, it doesn't appear at all, even as an unknown device.

samyk, 2016-12-22:
Are you using the right USB port? The one closer to the middle of the Pi0.

SpicyPiePete, 2016-12-23:
I am using the correct USB port. And, as for being able to ssh into the pi, I didn't think the target device could do that, unless PoisonTap had at least been able to initiate a connection to it. I'm going to go through #8 though, as you linked to in the other topic.

"Fiy on my rasbian when using the g_ether module, the configfs information in the startup.sh script is not taken into account and the usb gadget default to standard value, I have to load the libcomposite module in order for the configfs to be used."

I have a feeling this could be the issue. I'm going to try adding the libcomposite module to fresh install and see what happens.

SpicyPiePete, 2016-12-23:
I tried a fresh install adding libcomposite to the modules and had the same results, so I modified the pi_startup.sh script to allow the VID and PID changes. The device was now recognized by the windows machine, but as a group of WAN devices. The poisoned page still would not show up in a browser, and the server end is still only listening, but the log now shows the latest log listening followed by \00\00\00\00\00\00\00...