samyk/poisontap #80: Not giving Windows 10 an IP address

Adv3ntur3rz, 2016-12-18:
Everything is installed properly, latest node v6 is installed, and Windows 10 detects it as an ethernet device, but poisontap is not giving it an IP address to connect to. The log just says "poisontap is listening"

Any fix?

HanseSecure, 2016-12-20:
Which Windows Version do you use (Home, Pro, Enterprise)?
On Windows 10 Pro its working.

samyk, 2016-12-20:
Is dhcpd running on PoisonTap?

SpicyPiePete, 2016-12-22:
This seems to be the same problem I'm having.
HanseSecure: I was testing the device with Ubuntu, Win10 Home,and a few others. Since I wasn't getting any result, and since any that I did get could be easily cleaned, II went on a testing spree of plugging it into every computer I could for a couple days. Same result of nothing but the log listening

samyk: I might have a misunderstanding of the zero's power distribution abilities and such, but how can I watch for dhcpd to be running on Poisontap while simultaniously using it as intended?

samyk, 2016-12-22:
You can ssh into it or add serial over USB (try ssh'ing to 1.0.0.1 or raspberrypi.local). Also, try this: https://github.com/samyk/poisontap/issues/8#issuecomment-265818957

Are you using the right USB port? (the one closer to the middle, not the edge)

Adv3ntur3rz, 2016-12-26:
dhcpd is running,  but it says that it is not configured to listen on any interfaces. I did all of the proper commands and put dhcpd.conf in the right pace, so I am not sure where I went wrong.

samyk, 2016-12-26:
This is an RPi Zero? Paste the dhcpd error, config, and config location please.