samyk/poisontap #83: Backdoor.html not caching correctly.

dreadlocked, 2016-12-22:
Hi, I've configured Backdoor.html and target_injected_xhtmljs.html to connect my public server. The server is correctly running backend_server.js etc etc etc.

The problem is, when I disconnect the Raspberry Pi Zero from the USB port, and try to access, for example, www.nfl.com/PoisonTap, the page shows the personalized 404 error, this occurs whit all the websites, so backdoor.html simply doesn't load from anywhere it's not cached.

So my first question is, there is any way to solve this issue?
It's backdoor effectiveness dependent on the browser?

Thanks!

TimelineX, 2017-02-01:
I have the same problem

Jeremyyang920, 2017-02-24:
@dreadlocked and @TimelineX 
Did either of you manage to fix the issue? I'm having this issue as well....