samyk/poisontap #84: Wow, Poisontap open automatically a browser!

fEDUntu, 2016-12-22:
I'm doing a pentest for my company and in the meanwhile i'm trying to experiment some new physical attacks involving USB ports as attack vectors.
The workstation that I'm testing is a fully patched Windows 10 with all security updates, full-disk encryption and a strong "cloud-based" firewall.
I armed my Raspberry Pi Zero with PoisonTap correctly and, during my tests, in this particolar workstation **I surprisingly found that the PoisonTap is working without an open browser**, because after a minute Internet Explorer is automatically open with the incredible HTML5 canvas animation and suggest me that PoisonTap is working: cookies are stored in the log file, and the web-cache backdoors are reachable from my C&C Server! :)

I was surprised about that, and I started analysing the network traffic to find what services could trigger the PoisonTap. The only HTTP traffic that I found was direct to 3 IP addresses:

-  13.107.4.50 (Microsoft Corporation)
-  95.100.234.23 (Singapore Akamai Technologies)
-  23.12.108.239 (Paris Akamai Technologies)

I suppose this traffic is some sort of background Windows updates, and this triggered the PoisonTap. 
But what is currently not clear to me is **how this background traffic could pop-up a new Internet Explorer browser**?
Anyone else found similar behaviors?

clankill3r, 2016-12-26:
Makes me wonder if you could fake a windows update. Since windows so often installs software on a reboot...

wismna, 2017-01-04:
I noticed the same thing! It opened a new Edge window when no other browsers are opened, but a new tab in Firefox when it was opened.
I noticed nothing of the sort on either Mac or Linux.

samyk, 2017-01-04:
@clankill3r I doubt Windows Update would be susceptible as I'm assuming the updates themselves are signed, and likely the communication itself is protected, but there may be other vulnerabilities. You never know until you try!

mame82, 2017-02-22:
@fEDUntu 

In my opinion the answer to this magic is [th connectivity test](http://www.msftconnecttest.com/connecttest.txt) which is requested by win 10 every time a new network is detected.
The reason is that windows wants to tell you if your new network has Internet connectivity or if there is a captive portal involved and so on. Thus multiple steps are involved in doing this.

Checking the IP of www.msftconnecttest.com, you should see something familiar.
On win 7 this would be www.msftncsi.com.

Additionally Win 10 starts multiple HTTP requests to various Microsoft hosts, in order to send telemetry data.