samyk/poisontap #86: Raspberry Pi 3 breaks after reboot

gsakel1, 2017-01-03:
After following all the instructions for raspberry pi 3 and changing all instances of usb0 to eth0, the pi 3 "breaks" after reboot. None of the USB or ethernet ports function and I have to recover the OS when this happens (unplug the pi, plug it back in, hold shift to recover). The USB ports work on startup, but after it boots, none of the ethernet or USB ports work. Has anybody had this problem or know a fix for it?

Giusepp3, 2017-01-04:
I have the same problem, if i resolve i'll let you know

samyk, 2017-01-04:
I'm guessing part of the pi_startup.sh is causing issues, specifically anything USB related. Running on an RPi3 is actually a much simpler setup so I would just set it up manually (and I originally tested on an RPi3 + dongle with success).

I would cut out everything before the "ifup" command in pi_startup.sh, adjust all usb0 to eth0, but you will also need to adjust some configs as well (eg dhcpd.conf).

Giusepp3, 2017-01-05:
can you upload those file edited for the RPI3?

Giusepp3, 2017-01-05:
i've done this... please check if is correct
[pi_startup.sh.zip](https://github.com/samyk/poisontap/files/687190/pi_startup.sh.zip)

ZeusChops, 2017-01-11:
Start this process on crontab using `crontab -e` adding the location to your attached file on a new line. example:

`crontab -e`
```
~/Desktop/pi_startup.sh
```

Your method for this doesn't need to run every time, but the ethernet adapter settings are not being saved correctly on your build (nor are the last 4 lines which are specific processes which start once the startup file is run).

SpaceMan1228, 2017-01-25:
@samyk I don't think the problem is part of the pi_startup file. Because I booted the pi for the first time (after reformating and reinstalling the sd card. Because this prob has happened a lot b4). I installed the git for the poisontap. Did all the instructions. Except I didn't put pi_startup in rc.local, so the pi would boot up, but not do the pi startup file. But still it disabled the the eth and USB. Help !!

meatloaf18, 2017-11-01:
Was this resolved? I am experiencing the same issue. Poisontap is not working and the USB and ethernet ports do not work on the RPi. I cannot ssh into the pi because the ethernet is not working. I changed the usb0 to eth0, and I will also do what @samyk mentioned above once I get the usb/ ethernet working. Thanks!

samyk, 2017-11-01:
I haven't tested this myself on rpi3. It should work through the ethernet port, but as mentioned above, sounds like something specific to the pi0 is causing this issue.

Also, I suppose it depends how you're connecting the Pi to your network as the idea is that the Pi will be providing DHCP, not the other way around. You'd want to connect your Pi to your computer via USB-to-ethernet dongle and have your computer get an IP from the Pi (essentially backdoor yourself -- close your browser first to prevent that), then ssh to the Pi (your "gateway")

meatloaf18, 2017-11-01:
Thankyou for responding so quickly! Will plugging the raspberry pi into my computer with the usb to Ethernet adapter enable me to ssh into it? The Ethernet and USB ports are not currently working... then if I use the adapter with poisontap will I still have to change usb0 to eth0? Thankyou again.