samyk/poisontap #87: Windows 10 issues

wismna, 2017-01-04:
Hi,

First of all, thanks a lot for this amazing project! 
However, I couldn't for the life of me make it work on Windows 10. Even with the IBM 0x04b3 in idVendor and 0x4010 in idProduct, Windows correctly detected it but failed to load the driver with a code 10.

So, I investigated a lot and ended up adding a special RNDIS configuration first in the startup script. Windows now correctly loads the Raspberry Pi Zero (with no manual intervention!), and Linux ignores it to load the CDC ECM (now second) configration. 
MacOs should do the same in theory, but since version 10.11, it's been broken so you need to use exclusively the CDC ECM configuration.

samyk, 2017-01-04:
Awesome! Will you be able to submit a pull request?

wismna, 2017-01-04:
Sure, I can do that !

However, I should first warn you that in order to make that work, I had to make several other changes which you may or may not want to include in your original project. Creating another configuration adds another USB ethernet interface, so we end up with 2 interfaces, usb0 and usb1. The problem was that iptables, route and the servers (isc-dhcp, dnsspoof) are bound to usb0, so usb1 was incorrectly configured.

I didn't want to duplicate everything, so I created a bridge interface, br0, for all usbX interfaces, on which the routing, iptables rules and servers are set. The good thing is that we could add as many interfaces as we want, no further changes in configuration would be needed. We only need to install bridge-utils package.

That way, everything works as intended on Windows and Linux. Unfortunately, it doesn't anymore on MacOs, so this is a regression. I'll probably add a simple if/then in the startup script to deactivate the RNDIS configuration when wanted, so it'll work again on MacOs.

I uploaded my changes to a [repository I just created](https://github.com/wismna/raspberry-pi/tree/master/HackPi), you can have a look at it beforhand if you wish.

samyk, 2017-01-04:
Cool -- it would be interesting to add OS host detection (http://ix.cs.uoregon.edu/~butler/pubs/sadfe11.pdf) via USB fingerprinting, then produce the additional appropriate USB devices based on that.

wismna, 2017-01-05:
Indeed. I'll have a look at that.

domitor1927, 2017-01-08:
Hi, i have this problem to,
to test i just installed rndis driver on windows 10 and work ok the test, i will wait this change to implement in my raspberry

AddaxSoft, 2017-01-11:
I did the same work, and worked flawlessly on Windows 10 and 8, 8.1

I'm curious to hear if you got it to work silently on Windows 7

wismna, 2017-01-14:
Back with some news. After a lot (really a lot) of digging around, I managed to create a very basic OS fingerprinting mechanism. Now, when PoisonTap is connected in a MacOS computer, it will not load the RNDIS configuration.

So, everything works for me for all 3 majors OSs, Linux, Windows and MacOs, without any user interaction!
As the changes are quite extensive and even include a modified kernel module, I haven't submitted a pull request, but I have updated [my repository](https://github.com/wismna/raspberry-pi/tree/master/HackPi) with all the changes I made. 

Let me know what you think!

AddaxSoft, 2017-01-19:
great work! 
can you confirm this works on Win 7?

wismna, 2017-01-20:
Yes, it seems to be working on Windows 7 also :)