samyk/poisontap #88: How to forward all captured traffic to another NIC ? [Not an issue]

gsa390, 2017-01-05:
Hi,
I've just added a second ethernet port to the Pi Zero using the GPIO pins. Now I have two interfaces: usb0 and eth0 (connected to router).
I want that the victim machine continues to access to Internet, so I need to forward all packets coming from usb0 to eth0. 
How can I do?
I tried with iptables, enabling ip_forward in the system, but something is wrong and doesn't work.

Thanks

mame82, 2017-02-22:
@gdecostanzo 

My rough guess on this:

This [line](https://github.com/samyk/poisontap/blob/master/pi_startup.sh#L52) redirects traffic destinated to TCP port 80 of some foreign IP to **127.0.0.1**:1337. Where setting 127.0.0.1 is implied by the REDIRECT statement (see iptables manpage for details). So basically this DNAT to localhost.

What you need instead is MASQUERADING on your outbound interface like this:
`iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE ` while IPv4 forwarding has to be enabled with `/sbin/sysctl -w net.ipv4.ip_forward=1` .

Now my question is: Where could one find resources describing how to do this:

> I've just added a second ethernet port to the Pi Zero using the GPIO pins.

MarnixBouhuis, 2017-04-11:
@mame82 though my post is a bit late, here is how you do it: http://raspi.tv/2015/ethernet-on-pi-zero-how-to-put-an-ethernet-port-on-your-pi

mame82, 2017-04-11:
@MarnixBouhuis the for pointing out. Interesting read, diving into SPI is on my ToDo list. Anyway, switched to Pi Zero W in meantime, so Bluetooth and WiFi is available.
Could ETH over SPI reach 1 GBit/s ?

MarnixBouhuis, 2017-04-12:
@mame82 those spi network interfaces are most of the time capable of doing 10 Mbps. But the spi bus speed is a bottleneck, even if you increase the bus speed (spi-max-frequency = 20000000) you will get speeds around 400-500 Kbps. This is based on the Ethernet module I used (ENC28J60). You should always look at the frequency supported by the mic and pi when increasing the spi bus speed.

samyk, 2017-04-12:
You could use a USB hub on the OTG port to connect to a USB-to-Ethernet dongle and the target machine. The Pi's USB bus itself theoretically maxes out at 480Mbps, so a network card could get some decent speed, though not 1Gbps.

mame82, 2017-04-12:
@samyk Unfortunately using USB OTG or changing dtoverlay to sth else then DWC2 isn't an option for my current pawnage project, as I need USB gadget working.

https://github.com/mame82/P4wnP1

sssseossss, 2017-12-01:
Hi! 
Just to check if there is any way forwarding the traffic to wifi nic? Also does anyone knows why autoconnect to hotspot stopped working?
Thanks!