samyk/poisontap #90: (NOT AN ISSUE) GPIO to start poisonTap

fightforge, 2017-01-12:
Hi,
i'm using usbOTG to connect a wifi usb card to my network for simply use but if i configure interfaces file as tutorial, my raspberry cannot autoconnect to my access point. So i short with a jumper a gpio (gpio26 with gnd) to switch between poisontap device and normal wifi device. When jumper is connected to short gpio26 and gnd then poisonTap is started at boot. When jumper is removed then wifi is configured to connect to my access point.
My rc.local is shown below:

> echo "26" > /sys/class/gpio/export
> echo "out" > /sys/class/gpio/gpio26/direction
> echo "1" > /sys/class/gpio/gpio26/value
> sleep 1
> res=$(cat /sys/class/gpio/gpio26/value)
> sleep 1
> 
> if [ $res = '0' ] ; then
>  /bin/sh /home/pi/poisontap/pi_startup.sh
> else
>  ifconfig wlan0 up
>  wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf -B &
>  dhclient -nw wlan0
> fi

My wpa_supplicant.conf is:
> country=GB
> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
> update_config=1
> 
> network={
>         ssid="wifi_01"
>         psk="wpa pass"
> 
> }
> 
> network={
>         ssid="wifi_02"
>         psk="wpa pass"
> 
> }

if could be useful

darrylburke, 2017-01-12:
I do something similar but a lot more extensive.. I've used several jumpers across many pins to change the  Zero Pi into many HID devices (Storage, Keyboard, Ethernet, Serial, PoisionTap etc)

this way I can use it as a "multi tool" of several tools, not just PosionTap..  I have my code up at

https://github.com/darrylburke/RaspberryPiZero_HID_MultiTool

fightforge, 2017-01-12:
wow! thanks

sssseossss, 2017-12-01:
Nice! Thanks guys!