samyk/poisontap #92: PoinsonTap does not launch automatically

CoeurBis, 2017-01-17:
Hello,

I congratulate you on this work.

I only encounter a problem when booting: I plug the Raspberry Pi Zero with all the program installed, but nothing happens ...

I have to manually launch a page on the browser to launch the PoinsonTap.

Have I missed something? I repeated the procedures several times without success.

Thank you for your help

nonefaken, 2017-02-18:
Hi, the moment you plug Poisontap the computer must be running some webpage/s in web browser/s. As the webpage is opened in the background most of them will do periodic requests to Internet. This is what will trigger Poisontap, if it worked out. To find out if Poisontap works at all the easiest approach on my opinion is to check if you computer has received IP address from 1.0.0.0/0 network, for ex. 1.0.0.15. Check by typing in cmd "ipconfig /all".

CoeurBis, 2017-02-18:
I see, thank you for your reply