samyk/poisontap #99: Quick one about combo breaker

chefterry76, 2017-02-04:
Which code specifically should I rip out and if possible can you rip it out and re-post that version it would be soooo helpful thank you, and i know that this is the wrong git project but you aren't answering me. On the right git so please do it

t3chguy, 2017-02-04:
@chefterry76, I am not associated with either project but @samyk has 0 obligation to first answer you, second provide you help. Looking at the fact that the codebase for the project you are referring to hasn't been updated for 2 years maybe he has chosen to end support for it in its majority, maybe you are on your own, but you should never create Issues
+ on a different git repository just to try and get more attention
+ with no description and all the text in the title like you did over on samyk/combobreaker#8

You are merely spamming this project with multiple Issues childishly, 4 issues so far wrongly filed. If you cannot figure out his code then first learn Arduino basics to almost the point that you'd be able to write something similar (sans the algorithm), then try to apply his code to your project, if you cannot get it to work you should at least be able to get somewhere debugging it as you have learnt something by then, even if not you will at least be able to then create a *SINGLE* issue on the *CORRECT* repository with more information than just asking for a simpler example, explaining exactly, in technical depth why the code provided does not work for you and what happens instead of what should be happening.

_just my two cents_

chefterry76, 2017-02-04:
@t3chguy I am sorry if this is irritating to you and Samy but I am having trouble with finding which code to take out so that this will work with no encoder or servo it hasn't not been working and I have been trying to get help on the correct git but no one is answering if you or Samy could help me I would be very thankful

closed 2017-02-04 by samyk