samyk/poisontap #100: Client not opening a Websocket

damudd, 2017-02-12:
Hello,

first excuse my english please. Everything is working, except the part that the client has to open a Websocket. I replaced in bot filed YOUR.DOMAIN with my public IP (not the DNS Name). The Server is Responding to 1337, i can verrify that by testing it via devtools. The animation shows up, i can see in the Logs on the pi zero that he is collecting the cookies, but the Client wont open a websocket on 1337 after i plug out the pi! So far great Work!

Kind Regards

TB

LarsBehrens, 2017-02-17:
I've got to see i'm having exactly the same problem. My server is also listening on port 1337, i've edited the YOUR.DOMAIN, i can curl commands to port 1337 on my server (so the port is open for sure). But i simply cannot get a check-in from a client to my CnC.

samyk, 2017-02-17:
Does it attempt to in the console/network inspector?

nonefaken, 2017-02-18:
I suggest uncomment line "document.body.appendChild(iframe);" in file "target_injected_xhtmljs.html".
But be warned that it will result in session for every domain in yo configuration.

damudd, 2017-02-20:
Hi

tanks for you response.

"Does it attempt to in the console/network inspector?"

i checkt an i see that no websocket ist openning! but i can Curl as well. Anny suggestions?

"I suggest uncomment line "document.body.appendChild(iframe);" in file "target_injected_xhtmljs.html".
But be warned that it will result in session for every domain in yo configuration."

i will try that later 2 day.

kind regards 

Tilli

LarsBehrens, 2017-02-21:
Hi @samyk and @nonefaken,

First of all, thanks for the reply, I appreciate it. 

Unfortunately i haven't been able to get poisontap to open a ws to my CnC. I've created a tcp dump and saw that there is no outbound connection to my CnC IP.  (or whatever IP you put in backdoor.html). I also did try nonefaken's solution but this, unfortunately, did made any difference. 

Any clue what i'm doing wrong?

MarnixBouhuis, 2017-04-12:
@LarsBehrens does your browser support websockets? You can check this by going to [https://www.websocket.org/echo.html](https://www.websocket.org/echo.html). some older browsers cause issues. I'm currently working on an Ajax fallback for poison tap, maybe that will help.