samyk/poisontap #101: Backend Server/backdoor.html caching

Jeremyyang920, 2017-02-21:
Currently, I have my backend_server.js file running on one computer on port 1337. I have poison tap pluged into another computer on the same network and have changed all of the Your Domains to match that of the computer running the backend_server.

I can see a pending websocket under the developer tools in chrome so I know that it had opened an outbound websocket. After unplugging poison tap, I tried sending a curl command from the server and I can see that the request was sent, but nothing happened on the poisoned computer even though there is the websocket that is open. 

Is there something that I am doing wrong?

Jeremyyang920, 2017-02-21:
Update. So I am able to send curl commands and have it pop up on the poisoned computer. But after unplugging PT, the websocket closes and it seems nfl.com/poisontap was never cached properly.

Jeremyyang920, 2017-02-23:
@samyk Do you any idea why the backdoor is not properly caching? I'm able to see an outbound websocket, but nfl.com/poisontap never gets cached.

samyk, 2017-02-23:
Look at the inspector to see why it's not caching. Verify the headers when PoisonTap performs the attack (the caching) and then verify what the headers are when hitting nfl.com/poisontap -- Chrome gives good info about whether or not something is from cache or not

Jeremyyang920, 2017-02-23:
So looking at the headers, when I have PT plugged in, I went to nfl.com/poison tap, and this is what the header was.

Request URL:http://nfl.com/poisontap
Request Method:GET
Status Code:200 OK
Remote Address:1.0.0.1:80
Response Headers
view source
Access-Control-Allow-Origin:*
Cache-Control:public, max-age=99936000
Connection:keep-alive
Content-Type:text/html
Date:Thu, 23 Feb 2017 19:19:40 GMT
Expires:Sat, 26 Jul 2040 05:00:00 GMT
Last-Modified:Tue, 15 Nov 1994 12:45:26 GMT
Server:PoisonTap/1.0 SamyKamkar/0.1

KALRONG, 2017-06-02:
Im having this exact problem I have the same headers and did a test in chrome windows/linux and chromium in linux, all cases have the same cache issue. 

The only way I managed to get it to work was in Linux disconnecting the PT before the websocket timedout. This didn't work for Windows as the websocket aborts when the PT is removed and as there is no cache the backdoor never connects to the backend server.