samyk/poisontap #102: PoisonTap works on Windows 10 but doesnt on Mac Siara..

seiterseiter, 2017-02-22:
I did the normal setup but without the persistent backdoor but for some reason it wont show up as a Ethernet device on my mac but it does on my PC.

samyk, 2017-02-22:
Run `tail -f /var/log/syslog` while starting up PoisonTap on macOS and paste what occurs please.

samyk, 2017-02-22:
On the mac

seiterseiter, 2017-02-22:
It says "tail: /var/log/syslog: No such file or directory". If you meant system.log it doesnt say anything at all when I plug it in.

samyk, 2017-02-22:
Ah, sorry, meant system.log. Weird. Plug PoisonTap in, wait about three minutes, then run and paste the output of (from the mac): `/usr/sbin/system_profiler SPUSBDataType`

seiterseiter, 2017-02-22:
USB:

    USB 2.0 Bus:

      Host Controller Driver: AppleUSBEHCIPCI
      PCI Device ID: 0x1c2d 
      PCI Revision ID: 0x0005 
      PCI Vendor ID: 0x8086 

        FaceTime HD Camera (Built-in):

          Product ID: 0x8509
          Vendor ID: 0x05ac  (Apple Inc.)
          Version: 5.16
          Serial Number: CC2BB40QZ9DG6LL0
          Speed: Up to 480 Mb/sec
          Manufacturer: Apple Inc.
          Location ID: 0xfa200000 / 6
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
          Built-In: Yes

        Hub:

          Product ID: 0x2513
          Vendor ID: 0x0424  (SMSC)
          Version: b.b3
          Speed: Up to 480 Mb/sec
          Location ID: 0xfa100000 / 2
          Current Available (mA): 500
          Current Required (mA): 2
          Extra Operating Current (mA): 0
          Built-In: Yes

            Apple Internal Keyboard / Trackpad:

              Product ID: 0x0252
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 1.18
              Speed: Up to 12 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0xfa120000 / 4
              Current Available (mA): 500
              Current Required (mA): 40
              Extra Operating Current (mA): 0
              Built-In: Yes

            RNDIS/Ethernet Gadget:

              Product ID: 0xa4a2
              Vendor ID: 0x0525  (PLX Technology, Inc.)
              Version: 4.04
              Speed: Up to 480 Mb/sec
              Manufacturer: Linux 4.4.38+ with 20980000.usb
              Location ID: 0xfa130000 / 1
              Current Available (mA): 500
              Current Required (mA): 2
              Extra Operating Current (mA): 0

            BRCM20702 Hub:

              Product ID: 0x4500
              Vendor ID: 0x0a5c  (Broadcom Corp.)
              Version: 1.00
              Speed: Up to 12 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0xfa110000 / 3
              Current Available (mA): 500
              Current Required (mA): 94
              Extra Operating Current (mA): 0
              Built-In: Yes

                Bluetooth USB Host Controller:

                  Product ID: 0x821d
                  Vendor ID: 0x05ac  (Apple Inc.)
                  Version: 1.54
                  Speed: Up to 12 Mb/sec
                  Manufacturer: Apple Inc.
                  Location ID: 0xfa113000 / 5
                  Current Available (mA): 500
                  Current Required (mA): 0
                  Extra Operating Current (mA): 0
                  Built-In: Yes

    USB 2.0 Bus:

      Host Controller Driver: AppleUSBEHCIPCI
      PCI Device ID: 0x1c26 
      PCI Revision ID: 0x0005 
      PCI Vendor ID: 0x8086 

        Hub:

          Product ID: 0x2513
          Vendor ID: 0x0424  (SMSC)
          Version: b.b3
          Speed: Up to 480 Mb/sec
          Location ID: 0xfd100000 / 1
          Current Available (mA): 500
          Current Required (mA): 2
          Extra Operating Current (mA): 0
          Built-In: Yes

            IR Receiver:

              Product ID: 0x8242
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 0.16
              Speed: Up to 1.5 Mb/sec
              Manufacturer: Apple Computer, Inc.
              Location ID: 0xfd110000 / 2
              Current Available (mA): 500
              Current Required (mA): 100
              Extra Operating Current (mA): 0
              Built-In: Yes

samyk, 2017-02-22:
Well, the RNDIS/Ethernet Gadget is there. What do you see in the system Network preferences?

seiterseiter, 2017-02-22:
[image]

samyk, 2017-02-22:
Very strange. Are you a use that has the ability to become admin?

seiterseiter, 2017-02-22:
I have two accounts on my laptop, both are admin.

JetseVerschuren, 2017-04-21:
Apple didn't include drivers for RNDIS in mac.
So it will show up as connected device but mac doesn't know that it is a ethernet adapter.
https://en.wikipedia.org/wiki/RNDIS
> The Remote Network Driver Interface Specification (RNDIS) is a Microsoft proprietary protocol used mostly on top of USB.