samyk/poisontap #107: Logfile

jebusherr, 2017-03-02:
Hi there,

my PoisonTap installation and everything works fine.
But could someone help me with the creation of a logfile, please?

In target_backdoor.js there is a log.php mentioned, which is not included in this repo.
I've already read, that this is correct and okay. But I don't know how to create such a php file on my own. Or is it possible that PoisonTab writes a logfile without this script?
It should write the file on my apache2 webserver which is in the same subnet.

Thanks for your help.

samyk, 2017-03-02:
Technically you could grab the data out of your server's error log.

jebusherr, 2017-03-02:
Thanks for your fast response, Samy. 👍 
So I just change the src in target_backdoor.js to: http://MYDOMAIN/poisontap  ?
And then I can grab the data out of the apache error log or which error log did you mean?