samyk/poisontap #110: Adding additional domains? (e.g. Do you just edit getDoms()?)

victorhooi, 2017-04-14:
Is there a procedure for adding additional domains to the list of domains to backdoor?

I see there's a shell script `alexa1m.sh`:

https://github.com/samyk/poisontap/blob/8263c64c62fcc335e2df3f650dcbe490459d1e71/alexa1m.sh

However, is this script actually called from anywhere? I can't seem to find any mentions of it being called in the repo.

Also, in `target_injected_xhtmljs.html` there's a `getDoms()` function:

https://github.com/samyk/poisontap/blob/8263c64c62fcc335e2df3f650dcbe490459d1e71/target_injected_xhtmljs.html#L2578

Is it possible to simple append a domain to this function? Or do you also need to amend elsewhere?

samyk, 2017-04-14:
Just amend that one function. The alexa1m.sh is just shared to allow people to easily grab the list.

closed 2017-04-14 by samyk

victorhooi, 2017-04-14:
Awesome, thanks for quick reply!

Was finally going to get some time to play with the code this weekend.

Do you think it's worth documenting these sorts of questions (e.g. how to add domains, how to remove etc.) in a FAQ or wiki somewhere? (There doesn't seem to be a Github wiki for it yet). Or perhaps just append it to the README.md somewhere? Let me know if you have a preferred location and I'm happy to file a PR.

samyk, 2017-04-14:
Sweet. I think amending the README, perhaps with a FAQ, would be awesome!

victorhooi, 2017-04-14:
Done - https://github.com/samyk/poisontap/pull/111