samyk/poisontap #115: Is raspberry pi zero is necessary?

harshil2105, 2017-04-28:
Is raspberry pi zero only possible....can we not work with orange pi or any other pi

Cobchise, 2017-04-29:
Works like a charm on my Pi 3 Model B. Just had to make some adjustments changing references in some places from 'usb0' to 'eth0' and using the ethernet port to connect to target PCs (as discussed in Issue #6).

harshil2105, 2017-04-29:
How can we make changes?

sgript, 2017-07-30:
@Cobchise Hey, did you follow the instructions in [this](https://github.com/samyk/poisontap/issues/6#issuecomment-261601628) reply, regarding the ethernet-to-usb adapter?

As I've not got a Raspberry Pi Zero, but a Raspberry Pi 2 (Model B) which doesn't work. However, I rather invest in getting a Raspberry Pi 3 if it will work, as it's more useful rather than needing many adapters.

harshil2105, 2017-07-30:
I don't think it would work....it's needs to emulate Ethernet which I don't think rpi3 is capable of

sgript, 2017-07-30:
@harshil2105 From what @Cobchise is suggesting, he got the RPI3 working, with ethernet-to-USB adapters (though it's unclear, which is why I asked him), hence he changed `usb0` to `eth0`. 

It's worth mentioning @samyk has also said in 26dd781 that all Pi (1/2/3) should work, if used with the ethernet dongle. Although you would need to power the Pi separately as well, rather than trying to connect the micro-usb power to your laptop for example (which causes issues - stated [here](https://github.com/samyk/poisontap/issues/23#issuecomment-261715477) by someone who's tested it).

If this is the case, then I may be able to get this to work with the Pi 2 Model B as well. I've placed an order for a much needed ethernet-to-usb dongle.

harshil2105, 2017-07-30:
Main point of this task was to be covert I talked to sammy kamkar...when he
joined my discord server he said there is no point in making a project
bulky if it is going to be used covertly...

sgript, 2017-07-30:
@harshil2105 But that does not mean it will not work. The issue is not about being covert but rather whether these models will work or not. Clearly it has worked for @Cobchise using a RPI3 using the method suggested in the README.

harshil2105, 2017-07-30:
You could test it but I'm sure this would create problems

sgript, 2017-07-30:
In my viewpoint it's worth the effort, as although the Pi Zero is better at being covert and be that as it may, it does not provide a lot of flexibility for much else, needing many other adapters for peripherals. I rather try on existing hardware first, especially if it's worked for someone already without the Zero.

dernyn, 2017-09-01:
USB gadget mode is also available on other RPIs other than the Zero, the orange may too support it depending on the firmware, look into that or use the real ethernet card as suggested above. Like everything in life, you must experience it for yourself by try and error.

harshil2105, 2017-09-01:
Raspberry pi zero is the best in case of emulating etherner... whereas
orange pi is buggy and with very less support...in case of extensive
testing I saw orange pi sometimes work and sometimes it doesn't,I had the
same question why use Raspberry pi zero as they are not cheap and very
difficult to find..but I'm telling you Raspberry pi zero is the best choice
and I wouldn't recommend orange pi

meatloaf18, 2017-11-01:
@cobchise could you please explain how you did this? I have a usb to Ethernet adapter and I believe I have changed all usb0 to eth0 but I still cannot get it to work on Pi 2b. Furthermore, now the usb and Ethernet ports on my pi do not work. I have been searching found anybody that described how they got it to work on their Pi 2/3. Thanks!

dernyn, 2017-11-06:
the raspberry pi 2/3, etc. has a built-in usb hub in which the included network card is connected internally in-to... instead of using the usb connection per as otg, configure the dhcp service for the eth0....usb gadget mode otg with the pi 2/3 will not work....so eth0 is the only way..... configured the dhcp server for eth0 and the rpi eth0 manual IP network  configuration to assign your own ip address... when connecting to the pi use the raspberrypi.local or xxxxxx.local , xxxxxx is the name of the raspberry pi hostname , you need apples mdns running on the machine you will be doing the ssh from.

harshil2105, 2020-04-18:
Well who is still here ?? 
Long time no see!!!

dernyn, 2020-04-18:
I’m around...

PoisonCap159753, 2020-04-20:
Yes,because it's small