samyk/poisontap #118: error when i run pi_startup.sh ls: failed

theCake75, 2017-06-11:
error when i run pi_startup.sh
```
ln: failed to create symbolic link ‘configs/c.1/acm.usb0’: File exists
sh: echo: I/O error
sh: echo: I/O error
ln: failed to create symbolic link ‘configs/c.1/ecm.usb0’: File exists
ls: write error: Device or resource busy
ifup: interface usb0 already configured
SIOCADDRT: File exists
```
didnt modify the file
can you help me please ?

samyk, 2017-06-11:
I believe these errors are okay as PoisonTap just reattempts to setup the USB network interface each time it starts up.

closed 2017-06-11 by samyk

theCake75, 2017-06-11:
@samyk 
[image]

when I plug hdmi cable on pi zero i see that, and it don't move
freez on "errypi login:
net.ipv4.ip_forward = 1
"
no network detected on the target pc (windows 10)

thank for your reply !