samyk/poisontap #123: Pi Zero not recognized as ethernet controller

valentin8709, 2017-08-09:
Hello guys!
First, thank you for this tool, this is great and I love the idea! I tried to test poisontap on my Pi Zero, but it didn't work. Here is the step I did:
- downloaded poisontap... obviously
- updated dist and firmware on the Raspberry
- set a backend server listening on 1337 with the backdoor (working, I can connect with nc)
- set the poisontap following the exact instructions, on the github and checked with this video: https://www.youtube.com/watch?v=dYDa_VDbjPY
- so my dhcpd is set, everything is running smooth when I manually run the bash script, the IP is set correctly as the route table and everything is launched at startup.

But... my computers (macbook 2008, fedora 25 on MSI GS60 or Windows 8.1 on the same GS60) doesn't recognize anything. I tried to switch the idProduct and Vendor as described in the github, but nothing changed.

I must have missed something, do you know where? I am asking myself if I need a special USB cable (I use the one I use to transfer data on my smartphone).

Thank you for your help =)

samyk, 2017-08-09:
I would open up a tool like the Device Manager on Windows, run `lsusb` on Linux or run `system_profiler SPUSBDataType` on macOS and share the outputs so we can see whether it's at least coming up.

valentin8709, 2017-08-09:
Yep I should have done that before. Thanks for helping me. Unfortunately, there is no difference between a lsusb with Raspberry and one without Raspberry... 
- I checked the cable: I can transfer data on my smartphone
- I checked the USB port: ok too

Perhaps my Raspberry missed something? If I add a Wifi dongle, I can connect on my Pi with SSH, so both micro USB are working too...

samyk, 2017-08-09:
Anything happen in /var/log/{syslog,messages,system.log} when plugging it in? I would try another USB cable anyway.

valentin8709, 2017-08-09:
I tried with another cable, same behaviour. Nothing in the logs... Wouldn't be around the "thing" who turn the Raspberry into an ethernet controller? I just don't know how this part works, I guess this is these lines who enable it. 
```
echo "dtoverlay=dwc2" >> /boot/config.txt
echo -e "dwc2\ng_ether" >> /etc/modules
```

samyk, 2017-08-09:
Are you using the right USB port on the Zero?

valentin8709, 2017-08-09:
Yes I think, the one who's used in the videos / pictures, in the middle of the raspberry. It is also the one I use sometimes to plugg a wifi dongle, with a wlan0 interface configured in the /etc/network/interfaces file. But I never specified anything in the interfaces about the USB port. Could it be the problem?

valentin8709, 2017-08-10:
Alright I tried to comment the wlan0 section in /etc/network/interfaces. Now there is only loopback interface and usb0 interface, but same behaviour... I understand nothing =(

dernyn, 2017-09-01:
**Properly Enabling Gadget mode on the SDHC card with the proper Text Editor**
______________________________________________________________________________________________
You seem to be having a problem with the device not getting into USB Gadget mode, verify that you are not using the windows notepad.exe or a text editor that can not handle POSIX (new line or LF), etc. to edit the config.txt and the cmdline.txt files, this is a known issue within the LINUX or  RPI community when editing Unix/Linux files. So use Notepad ++ if it's a windows machine editing the files and go to the edit tab, go to EOL Conversion and select (Unix), and then edit the file. These two files may have to be replaced if the EOL conversion does not work to fix your previous editing with a non-LF text editor , and then you will have to redo the changes, your best bet is to backup your current progress with win32imager under windows, re-copy the files either from the .img file or from the web GitHub repo.

**Raspberry Pi Zero USB Functionality native HOST/GUEST modes**
______________________________________________________________________________________________
Pay close attention ....the firmware Self detects the USB modes on boot up. be aware!
The raspberry Pi Zero can be a USB host and a USB guest, but not both at the same time.
**host mode** allows USB hubs and single devices, etc. to be attached to the PI Zero.
**guest mode** is strictly for the gadget mode feature, in which it's a 1:1 direct connection with a computer, aka OTG mode. So if you have a USB Hub attached and booted already, you can not also go in OTG Mode by plugging in an OTG cable.

Raspberry Pi Zero OTG Mode/USB Gadget mode
______________________________________________________________________________________________
There is some background knowledge which kinda has to happen, mostly on the windows side of things if you are using Winblows.
-for some win. systems, an (Acer USB Ethernet/RNDIS Gadget) driver for the RPI Zero USB Gadget mode has to get downloaded and installed, I saw this on an old Instance of windows 10 32 bit(x86), 64 bit seems to download the driver, or natively has the driver and does this for you and installs it just fine. Check out=
(http://www.catalog.update.microsoft.com/Search.aspx?q=USB%20RNDIS%20Gadget) and (http://domotique.caron.ws/wp-content/uploads/telechargement/RPI%20Driver%20OTG.zip)

I encounter this very issue on an old laptop, this may be the case.
_____________________________________________________________________________________________________________
Samyk as I said on twitter, this network trick to highjack the network connections traffic is brilliant!
It's my favorite part of this implementation.

also, a good Idea is to have dd images for SD cards in the 8gb or 16gb flavors for people to simply dump the latest Raspbian vanilla with PoisonTap already implemented, I know this may be an image maintenance nightmare, but it's just a suggestion.
_____________________________________________________________________________________________________________

valentin8709, 2017-09-04:
Thank you very much for the information!! I will test it and give a feedback here =)

s0j0hn, 2017-10-08:
Same problem on my windows 10, raspberry pi zero recognised only as Usb serial com device. No way i can make it rndis ethernet gadget, any help ?

dernyn, 2017-10-09:
you need the driver I listed above.

s0j0hn, 2017-10-09:
even if it isn't acer laptop ?

dernyn, 2017-10-09:
it has nothing to do with acer... it just works with the hardware ID  of the usb network card emulation for usb gadget mode... trust what I say... the acer name is just a filler...so far this works really great with the driver

s0j0hn, 2017-10-09:
alright, i will try it this afternoon.
On the other had, does it mean that poisontap will not work if this driver is not installed on windows 10 ?
How can i know which host it will work ?

dernyn, 2017-10-09:
this seems to only happen on windows 7 32/64 and windows 10 with 32 bit

s0j0hn, 2017-10-09:
well mine is 64 bit windows 10 with all updates

dernyn, 2017-10-09:
I see... well basically it works on previous jessie version, something changed in the image which causes the hardware ID to not change, and windows may have some setting which should download the driver and installs it for you but it doesn't, I believe it's a userland setting which stops the automation, it depends on the prerogative of the windows user to download and install.

s0j0hn, 2017-10-09:
That's what i thought , i used raspbian stretch. I will try with jessie instead and will post the result here

dernyn, 2017-10-09:
it happened after the nov 25 jessie, I believe. I have to double check....

s0j0hn, 2017-10-09:
I just tested my raspberry pi zero v1.1 (added **modules-load=dwc2,g_ether** and **dtoverlay=dwc2
** respectively in correct files like tutoriels says) with
- 2016-05-10-raspbian-jessie-lite - does not boot up
- 2016-05-27-raspbian-jessie-lite - does not boot up
- 2016-09-23-raspbian-jessie-lite - does not boot up
- 2017-07-05-raspbian-jessie-lite - usb problem not recofnised as rndis cannot install it as network adapter

What i am doing wrong ? Is windows 10 not vulnerable to poisontap or something ?

s0j0hn, 2017-10-09:
It works with acer rndis drivers ... but at this point on ... i had to install them myself. This is pointless

dernyn, 2017-10-09:
it's not pointless, there may be a fix for the image and the poisontap startup script, just use the driver for now, the driver is just the generic usb driver, it doesn't actually installs any file, it just tells win10 to use its internal generic .sys file for rndis support for that particular hardware ID used by usb gadget mode, samyk used his own which was more common and didn't need a driver file pointing to the device ID. poke around learn and relax....tinkering is part of life.

valentin8709, 2017-11-05:
Hi again, 
Thank you all for your time. I tried many things but no result. I recently bought a Raspi Zero W (the first one was just a Raspi Zero) for a totally different project, but tried PoisonTap just for testing. And it worked! 

I don't understand why, but it works with my W version, and not with the simple version... So problem solved

dernyn, 2017-11-07:
did you place your startup script in /usr/bin ?

dernyn, 2017-11-07:
can you extract the image from the ad card and upload it?

valentin8709, 2017-12-05:
Hello @dernyn, sorry for the response time =/ I am checking the image, I upload it in the afternoon then give you the link here as soon as it's finished.

valentin8709, 2017-12-05:
Here it is, hoping it's not too late:
https://mega.nz/#!4Z911RJa!ECylHRHVWbqqaREqxYF4CPHK_H1bFJm4qjy1VJ8tPYw

dernyn, 2017-12-06:
Hey, @valentin8709 
Thank you very much for the upload.....I will see the differences to my configuration.
I will keep you posted. it's never too late...I always have time to test things like these.

P.S. compressing the .img file further into an archive format like .rar reduces the file size for web transfer.
turning that 7GB into 2.4GB..... I'm not being mean...just in case you didn't know. I'm grateful you shared your implementation. Thanks!

valentin8709, 2017-12-06:
Thank you @dernyn, indeed I totally forgot to compress the file. If I have time I will reupload a compressed .img for others who would like to download it.

valentin8709, 2017-12-06:
Here it is, the same image, but compressed:
https://mega.nz/#!NQtUFYjB!L4FgukQ7IOLEKsPHR4dL45LezY5mkLN5ioqr1xDkm80

Desc133, 2018-04-25:
Está funcionando com a última versão do raspbian no Raspberry PI Zero?

coramat, 2018-04-29:
@valentin8709 I've installed the image you've uploaded here but i'm having problems with the login :  the default credentials aren't correct and i'm not able to change the password

valentin8709, 2018-04-30:
Oups! I forgot to reset the password. The actual one is "Poison_Lock3d".

coramat, 2018-04-30:
@valentin8709 Thanks :)

dernyn, 2018-08-16:
@valentin8709 Sorry for the delay on my reply, the image you uploaded has the same problem, it won’t change the usb hardware to that of what Sammy configured on the script, I got it to do the Same as you did, but it won’t change IDs

morpheuslord, 2023-11-07:
For anyone having issues with RDNIS drivers I have listed the driver files and the usage procedures on this link : https://github.com/morpheuslord/ZERO_W_RNDIS_DRIVERS