samyk/poisontap #128: PoisonTap on Smartphone in futur ?

catest0, 2017-10-27:
A very powerful tool, would like to see it working on for example a Nexus 5... Thanks

simonpunk, 2017-12-25:
@catest0 I have got it working already using OnePlus5.

I think Nexus5 is not capable of emulating ECM Ethernet device as those old devices are using old usb gadget system. However, OnePlus5's usb gadget system is the same as pi zero, so if you also want to make it work on Mac OS, you have to choose a devices which using a new usb gadget system.

cegi, 2017-12-28:
@simonpunk Hi, can you explain how you made it work on your smartphone if you don't mind ?

simonpunk, 2017-12-29:
@cegi Hi, The most important thing is to fresh a fully working nethunter kernel and rewrite some codes based on the original script.

The nethunter Dev team just released out a semifinished kernel for oneplus5, and I just personally managed to finish the rest. It is kinda easy to make it work, but seems no one got time for that.

In the past, nethunter can emulate rndis protocol only, but now with ECM which works on MAC as well. I don't know whether other new devices are using the same USB system as Oneplus5, if so, it may work on those devices too.

For the script running on oneplus5, you may refer to this link 
[nethunter_oneplus5](https://drive.google.com/folderview?id=1D57nJDBirCuSvzFGqIuR5qDrpNNBpchc)

or refer to this [#942](https://github.com/offensive-security/kali-nethunter/issues/942)

cegi, 2017-12-29:
@simonpunk Thanks, i own a Nexus 5, with nethunter on it so i'll try, it might work as the Nexus 5 support ethernet USB so it should be able to emulate ECM ethernet. I don't know if it matters but Nexus 5 also has the same internet card as the Raspberry Zero and Pi 3.

simonpunk, 2017-12-30:
@cegi 
You can give it a try to emulate ECM. And I am meaning the scripts such as mubix-lock and poisontap script. But the poistontap script is still not fully working, I don't know why, maybe due to the browser update or OS update, I can't exploit it on both locked Mac or Windows. With unlocked Windows, it can be automatically poisoned as long as you have a browser and a http website running. With unlocked Mac OS, I need to refresh the http website to get it poisoned. So I am still looking for the answer..
maybe you can help to identify where the problems are.

cegi, 2018-01-01:
@simonpunk, i did not manage to make it work with Nethunter, i'm going to install Nexmon + Debian with linux deploy and try again, if i manage to make it work i will tell you if i can bypass lock.