samyk/poisontap #130: Pi Zero W - No Redirect to pages

himuura, 2017-11-16:
Poisontap installed on the Pi Zero W without issues, the .log is created but as soon as i connect it to the USB port, my internet connection is lost. Iptables is off (at least the service is). I think the problem is with the routes, i only have the 0.0.0.0 netmask 0.0.0.0 gw 0.0.0.0 U usb0. Will is stop work if i set the default GW as my wireless IP?

closed 2017-12-26 by himuura

samyk, 2017-12-26:
Was this resolved?

fedupwithhackers, 2018-02-28:
I was wondering the same thing.was it resolved?Amber.

himuura, 2018-02-28:
yup, it was. followed another install setup and it worked as intended. no internet when connected but the cookies were syphoned to poisontap. the things is, the websocket to the server is not working...the log is not created on the server side...already altered the backdoor.html file to the ip address of the server but still no connection is established, dunno why. @samyk , can you post a tutorial on how to install poisontap on the server side? We would be most thankful :D