samyk/poisontap #131: Autoconnect to Wifi stops working

sssseossss, 2017-12-01:
Hi!
Can you explain and suggest the solution on why the pi zero w stops connecting to wifi after launching the poisontap.
Thanks!

ge0rg, 2019-06-05:
Looks like some magic in raspbian prevents wifi from going up if you have usb0 configured.

Adding this to `/etc/network/interfaces` should help:

```
auto wlan0
allow-hotplug wlan0
iface wlan0 inet dhcp
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

MRtecno98, 2020-03-14:
> Looks like some magic in raspbian prevents wifi from going up if you have usb0 configured.
> 
> Adding this to `/etc/network/interfaces` should help:
> 
> ```
> auto wlan0
> allow-hotplug wlan0
> iface wlan0 inet dhcp
> wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
> ```

Does this broke poisontap? 'cause this seems like overriding nearly all the settings needed for poisontap to run.
(Sorry for any grammar errors, my native language is not english :P)